space-ros/space-ros #5: clang-tidy: non-const reference parameter

wjwwood, 2022-03-16:
migrated from https://github.com/osrf/sa-spaceros/issues/12 (originally by @mjeronimo)

clang-tidy identifies several occurrences of "google-runtime-references" where there is a non-const reference parameter and suggests to make it const or use a pointer. These occurrences in the Space ROS code (vnv list) should be evaluated and updated or identified as false positives.

ivanperez-keera, 2023-10-19:
We'll be planning milestones and releases differently, so I'm clearing the milestone for this issue.

ivanperez-keera, 2025-05-08:
This is has been stagnant for years. When we re-evaluate static analysis and code cleaning, this will come up again if it's still an issue.

closed 2025-05-08 by ivanperez-keera