space-ros/space-ros #7: clang-tidy: Default arguments

wjwwood, 2022-03-16:
migrated from https://github.com/osrf/sa-spaceros/issues/9 (originally by @mjeronimo)

clang-tidy complains about several occurrences of default arguments (google-default-arguments) used on virtual or override methods, which are prohibited according to the "google-default-arguments" check. We should evaluate each occurrence of the warning in the Space ROS code vnv-list and update the code accordingly or identify the occurrence as a false posititive.

ivanperez-keera, 2023-10-19:
We'll be planning milestones and releases differently, so I'm clearing the milestone for this issue.

ivanperez-keera, 2025-05-08:
This is has been stagnant for years. When we re-evaluate static analysis and code cleaning, this will come up again if it's still an issue.

@franklinselva are we ok to close this issue as not planned, or is this directly addressed by one of your current PRs?