space-ros/space-ros #9: clang-tidy: Explicit constructors

wjwwood, 2022-03-16:
migrate from https://github.com/osrf/sa-spaceros/issues/7 (originally by @mjeronimo)

clang-tidy complains about several occurrences of C++ constructors that could potentially be specified as 'explicit.'  For example,

```
"'operator Duration_' must be marked explicit to avoid unintentional implicit conversions [google-explicit-constructor]"
```

However, often there can be good reason for them not being explict. We should evaluate each occurrence of the "google-explicit-contructor" warning in the Space ROS code (vnv list) and update the code accordingly or identify the occurrence as a false posititive.

wjwwood, 2022-03-16:
migrated from a comment by me
Many of these are already marked `NOLINT` as they fall under the exceptions allowed by the Google style guide:

> Implicit conversions can sometimes be necessary and appropriate for types that are designed to be interchangeable, for example when objects of two types are just different representations of the same underlying value. In that case, contact your project leads to request a waiver of this rule.
> 
> Constructors that cannot be called with a single argument may omit explicit. Constructors that take a single std::initializer_list parameter should also omit explicit, in order to support copy-initialization (e.g., MyType m = {1, 2};).

-- https://google.github.io/styleguide/cppguide.html#Implicit_Conversions

In these cases our `NOLINT` are scoped with a `NOLINT(<violating rule to ignore)`, but the problem is that `cpplint.py` (which we run regularly) and `clang-tidy`, which we're adding, use different identifiers to show this, so I'm having to do some changes like this:

```diff
diff --git a/rclcpp/include/rclcpp/duration.hpp b/rclcpp/include/rclcpp/duration.hpp
index c6b790fc..6d562366 100644
--- a/rclcpp/include/rclcpp/duration.hpp
+++ b/rclcpp/include/rclcpp/duration.hpp
@@ -52,12 +52,14 @@ public:
   // intentionally not using explicit to create a conversion constructor
   template<class Rep, class Period>
   // cppcheck-suppress noExplicitConstructor
-  Duration(const std::chrono::duration<Rep, Period> & duration)  // NOLINT(runtime/explicit)
+  // NOLINTNEXTLINE(runtime/explicit, google-explicit-constructor)
+  Duration(const std::chrono::duration<Rep, Period> & duration)
   : Duration(std::chrono::duration_cast<std::chrono::nanoseconds>(duration))
   {}
 
   // cppcheck-suppress noExplicitConstructor
-  Duration(const builtin_interfaces::msg::Duration & duration_msg);  // NOLINT(runtime/explicit)
+  // NOLINTNEXTLINE(runtime/explicit, google-explicit-constructor)
+  Duration(const builtin_interfaces::msg::Duration & duration_msg);
 
   /// Time constructor
   /**
@@ -69,6 +71,7 @@ public:
 
   virtual ~Duration() = default;
 
+  // NOLINTNEXTLINE(google-explicit-constructor)
   operator builtin_interfaces::msg::Duration() const;
 
   // cppcheck-suppress operatorEq // this is a false positive from cppcheck

```

Which is fine, I just wanted to update this issue. A pull request is forth coming. I haven't worked my way through all of them yet, but so far they are all of this class of change.

wjwwood, 2022-03-16:
migrated from a comment by me

We decided to remove the Google checks: https://github.com/ament/ament_lint/pull/337

ivanperez-keera, 2023-10-19:
We are preparing an official release of Space ROS. This feature, already completed, will be considered part of that release, so it will be moved to the corresponding milestone.