space-ros/space-ros #29: Update the IKOS Installation to generate a .egg file

mjeronimo, 2022-03-30:
In the Dockerfile for Space ROS, the script is currently manually generating the .egg file from the egg-info directory. I would be better to fix this in the upstream repo and then update the Docker script accordingly. 

Here is the issue in the IKOS repo: https://github.com/NASA-SW-VnV/ikos/issues/185

ivanperez-keera, 2023-10-30:
I'm looking into this and also into changing ikos if needed.

A while back there was a change to the Space ROS `Dockerfile` to generate the egg file.

As far as i can see, that Dockerfile is gone, and I cannot find the steps that are currently used to install ikos (I see that `ros-dep` is used to install it, but I find no details of how it is installed).

Is this still an issue?

I used the following dockerfile to install IKOS on a fresh machine, and I cannot find any trace of an `*egg-info*` directory that should be packaged:
```Dockerfile
# Based on image by Maxime Arthaud
FROM debian:latest
MAINTAINER Ivan Perez <ivan.perezdominguez@nasa.gov>
ARG njobs=2
ARG build_type=Release

# Upgrade
RUN apt-get update
RUN apt-get upgrade -y

# Refresh cache
RUN apt-get update

# Install all dependencies
RUN apt-get install --yes gcc g++ cmake libgmp-dev libboost-dev libboost-filesystem-dev \
    libboost-thread-dev libboost-test-dev python3 python3-pygments libsqlite3-dev libtbb-dev \
    libz-dev libedit-dev llvm-14 llvm-14-dev llvm-14-tools clang-14 \
    git python3-distutils

# Add ikos source code
WORKDIR /root
RUN git clone https://github.com/NASA-SW-VnV/ikos

# Build ikos
RUN rm -rf /root/ikos/build && mkdir /root/ikos/build
WORKDIR /root/ikos/build
ENV MAKEFLAGS "-j$njobs"
RUN cmake \
        -DCMAKE_INSTALL_PREFIX="/opt/ikos" \
        -DCMAKE_BUILD_TYPE="$build_type" \
        -DLLVM_CONFIG_EXECUTABLE="/usr/lib/llvm-14/bin/llvm-config" \
        ..
RUN make
RUN make install

# Add ikos to the path
ENV PATH "/opt/ikos/bin:$PATH"

# Done
WORKDIR /
```

ivanperez-keera, 2023-10-31:
Tagging @nuclearsandwich because I suspect he may know too (sorry if that's not the case).

ivanperez-keera, 2023-11-04:
Circling back. I just sent a PR to ikos that switches to using `setuptools`. That creates an .egg file.

ivanperez-keera, 2023-11-04:
The PR in question has been merged in IKOS. It will be included in the next release.

I think this issue should be considered for the next release of Space ROS (Jan 2024), if still needed.

EzraBrooks, 2023-11-09:
I think we can keep tracking in #85 and just retitle it to be more general

ivanperez-keera, 2023-12-13:
The upcoming release of IKOS (3.2) addresses a number of installation issues. In particular, it uses Python 3, and venvs to install Python. I don't know if an `.egg` file will be necessary at that point.

If time allows it, I'd definitely like IKOS 3.2 to be included in Space ROS 2024.01.0.

ivanperez-keera, 2023-12-17:
The IKOS team has just made a release candidate available: https://github.com/NASA-SW-VnV/ikos/discussions/253.

I'd appreciate help setting up a test to determine 1) if that version installs correctly in space-ros, 2) if that version can be used by a demonstration application, 3) what needs to change in IKOS 3.2, if anything, prior to releasing.

ivanperez-keera, 2023-12-17:
I believe the current version of space-ros living in `main` will actually pick up IKOS 3.2-rc1, because we are not using version pinning, so it'll pick up IKOS from its master branch.

I could still use help figuring out how to exercise IKOS within space-ros, to make sure it's working correctly.

Tagging @mkhansenbot because this is also related to the demo we discussed.

ivanperez-keera, 2023-12-28:
IKOS 3.2 has now been released. The installation process has changed; a python egg may not be necessary.

I don't know if anything needs to change on the Space ROS side; seeing IKOS running in a Space ROS image, and further integrated with the dashboard would be the definitive piece of evidence we would need to determine that all is working as expected.

xfiderek, 2024-01-10:
> I'm looking into this and also into changing ikos if needed.
>
> A while back there was a change to the Space ROS `Dockerfile` to generate the egg file.
>
> As far as i can see, that Dockerfile is gone, [...]

From my investigation it looks like rosdep is not able to install ikos and it never gets installed into docker image. Please take a look at https://github.com/space-ros/docker/issues/104.

ivanperez-keera, 2024-01-13:
Since this is going to happen as part of space-ros/docker#104 and space-ros/docker#99, I don't think we need to keep this issue open. If something needs to happen as we update IKOS after each release, there'll be a separate issue specifically dealing with updating IKOS.

I'm closing this in favor of space-ros/docker#99, space-ros/docker#104 and #85. If you think some action remains as part of this issue, please re-open.

closed 2024-01-13 by ivanperez-keera