space-ros/space-ros #39: Review static analysis tools and their coverage

nuclearsandwich, 2022-04-29:
Upstream ROS 2 has disabled cppcheck >= 2.0 by default due to severe performance degradation with the way it is invoked by ament_cppcheck (https://github.com/ament/ament_lint/pull/345).

Before doing extra work to try and resolve the performance issues to preserve ament_cppcheck use for Space ROS, we should look at the checks cppcheck performs and determine how many are covered by our use of clang static analysis via clang-tidy, cpplint, and the rest of the ament_lint_common stack.

Once we have a comparison, we can propose either dropping cppcheck or investing time into investigating its performance issues.

ruffsl, 2023-03-17:
Looks like performance regressions are still a common topic for cppcheck tickets:

- https://trac.cppcheck.net/search?q=performance+regression

A recent example from three weeks ago:

- https://trac.cppcheck.net/ticket/11587

ivanperez-keera, 2024-02-01:
This issue is very open ended. There's no clear action item for this. It's also likely that part of this work will happen in the standards working group, where we evaluate how to comply with NPR7150.2.

I propose we either close this for now, or move this to a discussion if we want to keep this open for people to comment.