space-ros/space-ros #44: Enumerate all sources of memory allocation in the talker/listener demos

wjwwood, 2022-05-11:
In order to help us catalogue and address (whether that be by making sure its allocator can be overridden properly or by eliminating it for upstream ROS 2 or just Space ROS) all the sources of memory allocation in those two demos, we need to enumerate them first.

Use things like https://github.com/osrf/osrf_testing_tools_cpp#memory_tools to find all allocations and perhaps categorize them so we can reason about what needs to be done for each.

Definition of Done for this task is a list of all memory allocations for the `talker` and `listener` applications (C++) in the `demo_nodes_cpp` package, with optional categorization as it makes sense.

We'll use gcc and Ubuntu Linux 20.04 (maybe 22.04 @nuclearsandwich?) as that may affect what the std library will do based on its version.

Also, we could look at the call trace and check the API docs of all the system calls and std library calls made to ensure they shouldn't allocate memory in corner cases either.

nuclearsandwich, 2022-05-11:
> We'll use gcc and Ubuntu Linux 20.04 (maybe 22.04 @nuclearsandwich?) as that may affect what the std library will do based on its version.

We haven't yet migrated the Space ROS docker images to 22.04 nor the Space ROS repos file to the Humble release (instead of tracking Rolling development) but I think that forward planning should use 22.04 as the base. I will support that with image updates where I can.

closed 2024-06-24 by EzraBrooks

EzraBrooks, 2024-06-26:
Erroneously closed by an unrelated merge