space-ros/space-ros #56: ament_cppcheck does not use absolute filepaths for result reporting

Ronoman, 2022-08-05:
Problem is described and resolved in https://github.com/ament/ament_lint/pull/400. Documenting here as well since it is a problem that will impact SARIF processing, specifically determining if results across multiple static analyzers are pointing to the same file/line.

Ronoman, 2022-08-05:
This is going to be an issue with multiple of the static analyzer packages (ament_copyright is the next one I'm seeing). If there is a different solution to this problem (say, determine the absolute file path in post processing instead of during testing and replace SARIF) that is more desirable, I'm all for that as well.

mjeronimo, 2022-08-05:
Using an absolute path does tie the output to a specific machine. Perhaps there's a way to use relative pathnames in the SARIF files, but get the plugin to work by providing the base directory.

Ronoman, 2022-08-05:
Within the context of Space ROS, the path will be tied to the configuration of the docker image, not the developer's host machine. If everyone uses the same development environment using the provided docker images (which is probably a massive and incorrect assumption going into the future, but probably valid for now), then there shouldn't be differences between developers.

mjeronimo, 2022-08-05:
That's true. I'm OK with going this route for the Space ROS branch since we're not impacting anyone else. We'll start with the experience on Docker and then, eventually, Earthly or Dagger.

Ronoman, 2022-08-08:
The SARIF spec offers some guidance on this: https://docs.oasis-open.org/sarif/sarif/v2.1.0/csprd01/sarif-v2.1.0-csprd01.html#_Toc10540869

Copied again here: `If this artifactLocation object describes a top-level artifact and the value of its uri property (§3.4.3) is a relative reference, the artifactLocation object SHOULD contain a property named uriBaseId whose value is a string which indirectly specifies the absolute URI with respect to which that relative reference is interpreted. If the uri property contains an absolute URI, the uriBaseId property SHALL be absent. If this artifactLocation object describes a nested artifact, uriBaseId SHALL be absent.`

So, the `uri` property can be relative or absolute. If it is relative, then there should be a `uriBaseId` that helps locate the artifact within the filesystem. I see a couple of paths forward:

1. Require all static analyzers (really the ament wrapper) to report absolute URIs.
2. Require that the uri and optional uriBaseId produced by the static analyzer must create a valid absolute path.

I'll be opening more issues/PRs to some of the ament wrappers in the next few days, regarding these issues + some others I'm noticing with the current SARIF we are generating.

mjeronimo, 2022-08-08:
Thanks for the info. My inclination would be to use the second method, as it would seem a bit more flexible (perhaps for unanticipated usages that would require re-setting the base directory; don't know if that would happen in practice). 

I agree that we should make all of our SARIF output uniform, if possible and continue to improve the output (Cobra's in particular) so that it is more useful in environments such as VS Code.

Ronoman, 2022-08-12:
I've raised a PR into ament_lint: https://github.com/ament/ament_lint/pull/401

This adds `'uriBaseId': os.getcwd()` to the tools that are currently reporting `uri`s with non-absolute paths. I'm not sure I caught all of the packages that need the fix, but I (think I) fixed all of the ones that are currently not providing this information.

Ronoman, 2022-08-24:
Didn't pay enough attention while implementing this the first time around and included a breaking bug. Raised a new PR in ament_lint that resolves these issues: https://github.com/ament/ament_lint/pull/404

Once this PR is merged in, I believe this issue can be closed. Only remaining work would be if I missed an `ament_` package that isn't currently writing absolute paths.

mjeronimo, 2022-11-18:
@Ronoman  Looks like this one can be closed?

Ronoman, 2022-11-18:
Yep! I will re-open if I find that VSCode can't link back to source files from within the SARIF viewer, but I haven't seen this bug in a while so I'm comfortable to close.

closed 2022-11-18 by Ronoman