space-ros/space-ros #57: Not all static analyzers are run for all packages

Ronoman, 2022-08-09:
If I'm not mistaken (which is quite possible, new to ROS2), the only tests run for a package under `colcon test` are ones that are listed in that package's `package.xml` within `<test_depend>package_name</test_depend>` tags. Some ROS2 packages (take `rcutils` for example) test_depend on ament_lint_common, which in turn depend on all of our static analyzers (ament_clang_tidy, ament_copyright, ament_cobra, ament_cppcheck, etc). However, other packages (take `tf2` for example) only depend on a subset of these analyzers, so they are not all run.

I'm not sure what the best way to resolve this is. There are two things I can think of:

1. If the only place to indicate test dependencies is in a packages `package.xml` (which I believe is the case, but again could be wrong),  insert `<test_depend>` tags during `docker build`, or when another common command is triggered (`colcon build`, `colcon test`, `entrypoint.sh` for the docker image, or maybe an explicit command run by the user).
2. Upstream the <test_depend> to all of the ROS packages we want to run our static analysis tools on, either to the main ROS2 branches, or to spaceros branches.

nuclearsandwich, 2022-08-15:
The Space ROS project has been doing currently relies on the fact that many ROS 2 core packages rely on a metapackage called ament_lint_common, which is for the "common" set of linters and static analysis tools used by ROS 2 core packages, and patching ament_lint_common in Space ROS to include additional static analysis tools (or their ament / ament_cmake wrappers). Here's the [current list of "common" linters from ament_lint's spaceros branch](https://github.com/ament/ament_lint/blob/b27a723ffe6ec3970700afea7a613e3111309bab/ament_lint_common/package.xml#L18-L29).

This has the advantage of being relatively low touch and only requiring changes in one repository but the disadvantages are that it relies on all packages being considered for static analysis depending on ament_lint_common rather than an explicit list of analysis tools. Since the listed linters are expressed as dependencies that means that changing the list must be something happens in source at build time rather than during some later operation, especially since the CMake hooks which add tool runs to the test processes need to be updated when changing these dependencies.

I'm just getting back from extended vacation but somewhere I've got the start of a proposal to try and decouple the build-time dependencies from the test-time static analysis runs. However doing so isn't trivial and there are several considerations and trade-offs with moving to that approach. I'll try to get that proposal resurrected and share it for feedback both within the ROS team and Space ROS itself.

ivanperez-keera, 2024-01-11:
We'll revisit which static analyzers are executed for each package. However, this issue has no action item or taker attached to it, and it's too big to be addressed. I'm closing this and we'll revisit in the future.

Thanks a lot, @Ronoman :smile:

xfiderek, 2024-03-13:
PRs raised. they need to be merged in the following order:
ament_lint - https://github.com/ament/ament_lint/pull/482
spaceros - https://github.com/space-ros/space-ros/pull/145
docker - https://github.com/space-ros/docker/pull/141

asimonov, 2024-09-09:
@xfiderek I see you have some understanding of how our static analyzers are run. can you please comment on my question of upgrade to jazzy in relation to ament_lint status? https://github.com/space-ros/space-ros/pull/205#issuecomment-2337998327

commit 08a6e0ae (2025-04-03): Merge branch 'lint_all_packages' into main. Close #57.
closed 2025-04-03 by ivanperez-keera