space-ros/space-ros #58: Add SARIF post-processing capabilities to Space ROS

Ronoman, 2022-08-12:
Currently, many of the static analysis tools output their results in the SARIF format. While this is great as a common interchange format for these tools, extra processing needs to be done on these files afterwards to turn them into actionable development tasks for contributors.

I've been working on a set of SARIF processing tools, built in python, available here: https://github.com/space-ros/process_sarif. Below I'll list the features currently available in these tools, and the features I'm looking to add. I hope to use this issue as a spot for feedback from other developers on the kinds of post-processing we want, to help guide the development and integration of this tool into the standard developer workflow. If there's a different place where this discussion should take place, let me know.

Once this tool is more robust and ready to use, I can do a brief demo for the Space ROS ACO group.

Features:
- A python object model for interesting components present in SARIF files.
    - This object model is available in [process_sarif/sarif.py](https://github.com/space-ros/process_sarif/blob/main/process_sarif/sarif.py). It currently implements SarifFiles, Results, Artifacts, Tools, as well as a few sub-fields. `sarif.py` has documentation for each of these objects, and is capable of constructing these objects from JSON directly. Currently it only implements a small subset of the SARIF spec, but it is easy to extend these objects with new fields and relationships if desired.

- Helper functions to manipulate these objects.
    - These helper functions are available in [process_sarif/sarif_helpers.py](https://github.com/space-ros/process_sarif/blob/main/process_sarif/sarif_helpers.py). They consist of functions that either I am using often while developing and testing the tool, or functions that I anticipate others will want to call regularly. It currently has methods to load in all SARIF from `build/**` (`get_sarif_in_build()`), replace missing MISRA rule text (`replace_misra_rules()`, though this should probably be moved to static analysis tooling, instead of living in post-processing), and finding Results that may be duplicated of each other (`find_duplicate_results()`).

- A few tools documented in the [README](https://github.com/space-ros/process_sarif#1-tools).
    - These tools are scripts that call functions in sarif.py, sarif_helpers.py, and add small bits of functionality to provide useful output to the developer.

Desired features:
- Aggregation of Results by ROS package, static analysis tool, source file, and level (error, warning, etc)
    - This will help us determine what packages/files need the most work, and sort by severity.
- Integration with the Space ROS dashboard
    - Ideally, the output from these tools would be available in the Space ROS dashboard, without requiring developers to run them manually. I'm not sure how to pass information from these python tools to a plugin running in VS Code, but that would be ideal.
- Track Result deltas over "time" (test runs)
    - We want a way to determine how the number and location of Results is changing over time. This may be a part of the Space ROS dashboard already, I'm not sure. This would require this tool to be stateful across runs, and be able to match Results across runs of the static analyzers.
- Stretch: Integrate these tools into CI, and have the output available on Github Pages (or other documentation)
    - It would be great to automatically run these tools (along with test output) automatically when changes are pushed to Space ROS repositories. The results, after post-processing and visualization, could then be displayed in public-facing documentation, hopefully encouraging open source contribution.

Sorry for the wall of text!

mjeronimo, 2022-08-15:
Regarding tracking deltas over time, I'm thinking about how we achieve persistence. One way would be to introduce a database such as sqlite3. This would require converting (at least a good chunk of) the SARIF format to a database schema. Benefits would include persistence and the ability to easily perform arbitrary queries over the data. In this scenario, we'd still want to keep the original SARIF files (perhaps also in the database). An alternative is to keep the SARIF files and use them as the input, creating, data structures in memory and then writing code for each query required. 

Also, another desired feature is removing duplication of reported issues, both within the output of individual tools and across tools.

mayman99, 2022-08-20:
Great initiative, I like the idea of this tool, imho having a tool to post process the `sarif`s before they are fed to the dashboard could be a very convenient way to solve:
- Issues duplication across tools.
- Persistence. persistence currently works with the dashboard manually, like you manually put different build results in different folders and tell the tool which is which, I think this tool would be perfect to post process the results and put them in the format that works with the dashboard (or maybe we explore using a db instead of just folders).

Some of the features are you listed are already in the dashboard, like the aggregation of results, also track Result deltas over "time" to some degree as it only takes the issues count into account  ... etc

Ronoman, 2022-08-26:
New features to add to post-processing after our discussion yesterday (8/25):
- Export lossy (but correct) post-processed SARIF to a tarball
- Implement a sane way to invoke post processing
    - Could be either a new colcon verb (more correct but high overhead), or a script (python/bash/other) invoked at some common step

mjeronimo, 2022-08-26:
I like the colcon verb approach. I don't think the overhead will be an issue. The basic flow is:

```
colcon build    # generate binaries
colcon test     # run tests, generating JUnit XML and SARIF output
colcon sarif    # process the SARIF files, generating the tarball which includes the original SARIF files as well as the filtered output (removing duplication). 
```
The tarball is then the input for the Dashboard, where the user can:
* Peruse the data via multiple views
* Create logical groups of issues (for example, all preprocessor-related issues)
* Perhaps assign priorities to issues and groups of issues
* Disposition issues (indicate "Not an Issues", for example).
* etc.

The Dashboard will have to create its own persistent data to do some of these items, like grouping and dispositioning.

mjeronimo, 2022-08-26:
FYI - I'm working on a "colcon sarif" prototype that layers on top of your existing code. If we're happy with it we can discuss integration of the two.

Ronoman, 2022-08-26:
The overhead I was referring to was development time, as I'm not at all familiar with colcon. I agree it's probably the better way forward, and if you can get it working that would work great.

mjeronimo, 2022-08-26:
Oh, OK. Yes, I've got it working and have interfaced to a couple of the functions (main routines) in process_sarif. We should discuss what the colcon interface should be that we present to the user. I've started from the colcon-test-result plugin.

Ronoman, 2022-12-09:
I think I'm happy to close this, we have post processing implemented to the extent we need right now. Still some small changes needed, but no longer a big enough issue to keep this up. Thanks for the bump