space-ros/space-ros #62: control_msgs needed to run demos

ana-GT, 2022-09-28:
Hi. I built a space-ros workspace and added the **demos** repository (with mars and canadaarm packages). This build failed because both packages depend on **control_msgs** , which does not seem to appear in ros2.repos.

mkhansenbot, 2023-03-08:
When I run the docker container for the space_robots demo, it generates this repos file: 
```
spaceros-user@30a98981eed0:~/demos_ws$ cat /tmp/demo_generated_pkgs.repos 
- git:
    local-name: backward_ros
    uri: https://github.com/ros2-gbp/backward_ros-release.git
    version: release/humble/backward_ros/1.0.2-3
- git:
    local-name: control_msgs
    uri: https://github.com/ros2-gbp/control_msgs-release.git
    version: release/humble/control_msgs/4.1.0-1
- git:
    local-name: diagnostics/diagnostic_updater
    uri: https://github.com/ros2-gbp/diagnostics-release.git
    version: release/humble/diagnostic_updater/3.1.0-2
- git:
    local-name: realtime_tools
    uri: https://github.com/ros2-gbp/realtime_tools-release.git
    version: release/humble/realtime_tools/2.5.0-1
- git:
    local-name: ros2_control/controller_interface
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/controller_interface/2.22.0-1
- git:
    local-name: ros2_control/controller_manager
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/controller_manager/2.22.0-1
- git:
    local-name: ros2_control/controller_manager_msgs
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/controller_manager_msgs/2.22.0-1
- git:
    local-name: ros2_control/hardware_interface
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/hardware_interface/2.22.0-1
- git:
    local-name: ros2_control/joint_limits
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/joint_limits/2.22.0-1
- git:
    local-name: ros2_control/ros2_control
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/ros2_control/2.22.0-1
- git:
    local-name: ros2_control/ros2_control_test_assets
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/ros2_control_test_assets/2.22.0-1
- git:
    local-name: ros2_control/ros2controlcli
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/ros2controlcli/2.22.0-1
- git:
    local-name: ros2_control/transmission_interface
    uri: https://github.com/ros2-gbp/ros2_control-release.git
    version: release/humble/transmission_interface/2.22.0-1
- git:
    local-name: xacro
    uri: https://github.com/ros2-gbp/xacro-release.git
    version: release/humble/xacro/2.0.8-1
```

Should all of these be added here or there?

ivanperez-keera, 2023-11-17:
Related to #98 .

ivanperez-keera, 2024-01-12:
@ana-GT This seems to be working on my side.

Could you please check if this is still an issue?

Right now, what I see in a fresh `space_robots` image is that the `moveit2_generated_pkgs.repos` includes `control_msgs`. Should that be enough to satisfy this dependency, since `space_robots` is built on top of `moveit2`?

mkhansenbot, 2024-01-26:
I think we can close this. I'm guessing the author didn't run 'rosdep install' before trying to build the demos. Also there's been no comment follow up in 16 months, and our CI (which does run rosdep) works fine.

closed 2024-01-26 by mkhansenbot