space-ros/space-ros #67: Add Nav2 to Space ROS

mkhansenbot, 2023-03-15:
## Feature request
I'm planning to add Nav2 to Space ROS for use in Mars rover demo and other ground rovers
#### Feature description

Adding Nav2 will enable Navigation in Space ROS
#### Implementation considerations

I plan to implement this as follows:

- Add Nav2 to repos file, get it building in the Earthfile
- Add Nav2 to demo
- Run static analyzers on Nav2 and fix issues

mkhansenbot, 2023-08-26:
The first PR for this is: https://github.com/space-ros/docker/pull/80

mkhansenbot, 2023-12-31:
These are the minimum steps needed:

- [ ] Add odometry to curiosity mars rover gazebo model
- [ ] Add lidar to curiosity mars rover gazebo model
- [ ] Add Gazebo to ROS bridge to mars rover demo launch file
- [ ] Add odometry TF publishing to mars rover demo
- [ ] Add launching of Nav2 and Slam toolbox to mars rover demo
- [ ] Configure Nav2 parameters
- [ ] Add test to GH actions build flow

mkhansenbot, 2024-01-21:
This is now in review in these PRs: 
https://github.com/space-ros/simulation/pull/10
https://github.com/space-ros/demos/pull/22
https://github.com/space-ros/docker/pull/80

mkhansenbot, 2024-01-29:
Added an issue to track the need for a test: https://github.com/space-ros/docker/issues/125

ivanperez-keera, 2024-02-01:
@mkhansenbot since you have an issue to track the tests, and you completed most other things (AFAIK), can we close this parent issue?

closed 2024-02-20 by mkhansenbot