space-ros/space-ros #69: Set up Space ROS technical meetings

EzraBrooks, 2023-05-29:
(not sure if there's a better repo in the org to put non-code-related Issues on, let me know if there is @mjeronimo @nuclearsandwich)

We're aiming to set up a recurring meeting with technical members of the committee to be able to focus on technical implementation as opposed to the normal committee meetings that are, by necessity, going to be primarily strategy-oriented.

Please let me know on this issue whether you are interested in being part of this group. Once we have an initial invitee list we'll kick things off.

mjeronimo, 2023-05-29:
@EzraBrooks Thanks for setting this up. I'm interested to attend (and I think this repo is fine for this).

gbiggs, 2023-05-29:
I'd like to attend.

quarkytale, 2023-05-30:
I'd like to attend too!

mkhansenbot, 2023-05-30:
yes please

skolk, 2023-05-30:
I'll send people to this issue when they want to be included.

EzraBrooks, 2023-06-01:
Thanks everyone who's responded so far! I'll circle back on Monday to poll about a time for the first meeting.

salvioli, 2023-06-05:
I am interested to attend as well.

anmavrid, 2023-06-05:
I'd like to attend

EzraBrooks, 2023-06-05:
Hi all, I've just set up a Calendly poll for some times this week and next. I figured morning here in mountain time makes the most sense to accommodate eastern US and Europe timezones.

https://calendly.com/d/2kk-7rs-bjn/space-ros-technical-meeting

EzraBrooks, 2023-06-06:
I've sent out an invite for the most-voted-for time slot. This Thursday, Jun 8, 9:30-10:00 mountain time. See you then!

closed 2023-06-06 by EzraBrooks