space-ros/space-ros #71: Next Space ROS Technical Meeting agenda (date TBD)

EzraBrooks, 2023-06-08:
# Agenda Items:
- [ ] Introductions of new attendees
- [ ] #70

EzraBrooks, 2023-06-08:
Feel free to discuss in this thread thoughts on what we should cover next meeting, and we can add them as subtasks here.

mkhansenbot, 2023-06-08:
- [ ] Projects we want to prioritize (ie Astrobee?)

- [ ] Sync and release strategy #72

mkhansenbot, 2023-07-06:
When's the next meeting?

EzraBrooks, 2023-07-06:
Hey @mkhansenbot, sorry for the radio silence, I've been working on getting PickNik to sponsor a shared Slack channel so we can collaborate in real time. That should be happening very soon (ideally before end of week), and we can talk there once it's up. Personally I'm super swamped for the next couple weeks, so we can either meet in a few weeks or I can have someone else like @eholum represent PickNik at the meeting.

roncapat, 2023-07-10:
Minor points I would like to add:

[Add brief introduction to the project on organization repo](https://github.com/space-ros/.github/pull/2)
[How to proceed with code conformance fixes upstream](https://github.com/space-ros/space-ros/issues/73)

EzraBrooks, 2023-07-25:
Calendly for next meeting: we talked in Slack about it being this Thursday. https://calendly.com/d/4js-zyt-s3f/space-ros-technical-meeting

mkhansenbot, 2023-07-26:
The poll has ended, but I'm fine with Thursday

ivanperez-keera, 2023-07-26:
I've developed a process that I want to propose we follow for Space ROS. It would help us comply with some of the requirements that we'll have if we really want to deploy any of this. I'd like to have the floor for a good 5 minutes (ideally even 10).

ivanperez-keera, 2023-07-26:
Tagging @EzraBrooks for awareness.

mkhansenbot, 2023-07-26:
@EzraBrooks - I actually can't make the meeting at 11:00 PST, Can we pull it in to 7:00 or 8:00 instead?

ivanperez-keera, 2023-07-26:
7am and 8am are pretty bad times for me. It's when I commute to the office.

I'd suggest in general not scheduling meetings until 9am. People who don't work from home or need to take kids to school may find it hard to attend meetings at those times (especially on short notice like this).

closed 2023-07-27 by EzraBrooks