space-ros/space-ros #75: Space ROS Technical Meeting Agenda (Date TBD)

EzraBrooks, 2023-07-27:
Meeting time poll: https://calendly.com/d/4h7-29d-k6x/space-ros-technical-meeting

## Agenda
- [ ] Introduce any new attendees

mkhansenbot, 2023-08-10:
More topics:
 - can we move this communication, polls, etc to github discussions?
 - now that we have a working build again, what issues are next in priority?

ivanperez-keera, 2023-08-10:
I propose that proper issues be created only for things that would lead to changes in the repository's code. For anything that will not lead to a change in the repo, we can either use discussions, or we can create a separate repo that is only for coordination (I've used both in the past, there are pros and cons).

EzraBrooks, 2023-08-10:
Next meeting: #77 (we're giving discussions a shot for organizing these conversations)

closed 2023-08-10 by EzraBrooks

ivanperez-keera, 2023-08-10:
Regarding discussion templates: https://github.blog/2023-01-09-github-discussions-just-got-better-with-category-forms/