space-ros/space-ros #79: Proposed release milestones with dates

mkhansenbot, 2023-08-24:
From our meeting discussion #77 - we are proposing to have milestone releases every 3 months. In order to sync with upstream ROS 2 releases, we would release one month after the ROS 2 release.

April - Ubuntu release
May / June - ROS 2 release
July - Space ROS release

Then every 3 months after, so the annual release cycle would be :
Jan, April, July, October
(end of month)

Please comment or give thumbs up to agree

mkhansenbot, 2023-08-26:
This should address #71

ivanperez-keera, 2023-10-03:
We've discussed this before but I agree with this calendar. I'd rather fix the date as precisely as we can (e.g., the last day of the month that is not a holiday/weekend).

EzraBrooks, 2023-10-05:
We're going to try to release by EOM this month, October 2023, and proceed on the 3-month cadence. @Bckempa has volunteered some time (thank you!)

Various notes from discussion:

Release naming:

datestamp-distro? `2023.10-humble`
distro-datestamp-version? `humble-2023.10.0`

Tasks that need to be accomplished within the window of ROS 2 release -> Space ROS release:

- [ ] Only release if the new ROS 2 release is LTS
- [ ] Create release/distro branches on docker, demos, docs, and process_sarif
- [ ] Update the .repos files to the new versions for the new distro sync
- [ ] re-run the tests, creating issues for any that failed
- [ ] Docker images are already built but only tagged `latest`, must tag release as release name

closed 2023-10-05 by EzraBrooks