space-ros/space-ros #89: README link to contributing rules broken

mkhansenbot, 2023-10-04:
Of course, as soon as I merged PR #87 I checked the link on the README and realized it is broken:
https://github.com/space-ros/space-ros/blob/3418a46c3380e5250ad1185808a6295c4d306209/README.md?plain=1#L7

The actual file name is CONTRIBUTING.md (note the all caps).

I'll submit a PR momentarily to fix.

ivanperez-keera, 2023-10-04:
I'm a bit confused. I noticed the issue in the current HEAD.

I missed this issue and didn't see a PR, so opened issue #91 and sent PR #92.

If yours is ready though, you can send it instead. As you prefer.

ivanperez-keera, 2023-10-04:
I closed "my" issue as a duplicate of this one and re-sent my PR to use this issue number instead.

commit 94740a52 (2023-10-05): Merge pull request #92 from ivanperez-keera/readme-fix-linkcontrib

Fix link to contributing guide (Close #89).
closed 2023-10-05 by EzraBrooks