space-ros/space-ros #90: Add markdown lint check to space-ros/space-ros

mkhansenbot, 2023-10-04:
## Feature request
Issue #89 showed that we don't have a markdown link check in our CI for this repo (space-ros/space-ros). 

#### Feature description

We have added it in the `docker` repo: https://github.com/space-ros/docker/pull/78

#### Implementation considerations

We really need to add it to every repo, since they all will contain markdown files.

ivanperez-keera, 2023-10-04:
Is this issue just a matter of adding this file to this REPO?

https://github.com/space-ros/docker/blob/main/.github/workflows/markdown-link-check.yml

ivanperez-keera, 2023-12-03:
@Bckempa Do we really have the resources to commit to addressing this in the next release too (has someone accepted that they'd finish it before the next release)? If unsure, we should leave it out of the release for now.

ivanperez-keera, 2023-12-03:
I see.

I think we should prevent the case where we don't complete everything we plan for one milestone and we start carrying issues forward on. At that point, milestones and planning lose their meaning, we'll have a constant feeling of disappointment, and we'll likely end up in a situation similar to the one we were in 2 months ago. That's why I'm being so conservative (also with the ones **I** commit to).

If you are confident that you'll have the cycles to fix it when you do that pass before the next release, I'd say go ahead and assign this one to yourself.

If you are not sure yet --especially with xmas in the middle--, I'd strongly suggest that we aim for less than we think we can complete (that is, we leave ample margin of error). We can always assign a few more later on if we finish earlier than expected -- but descoping IMO should be avoided.