space-ros/space-ros #155: 2024-07 Release

mkhansenbot, 2024-05-01:
Based on last release flow: #153 
Before the release window

- [ ] Determine the release tag of the form rosdist-year.month.point, for example humble-2024.01.0
- [ ] Open a new issue with a title of the form "year-month Release", for example "2024-01 Release"
- [ ] Create a milestone that matches the release tag in each repo under the Space ROS organization
- [ ] Create a new view in the Space ROS Project filtered by milestone

During the release window

- [ ] Tag issues intended for the release with the correct milestone
- [ ] Update issue status on the project view for the release

At the end of the release window

- [ ] Create release/distro branches on docker, demos, simulation, docs, and process_sarif.
- [ ] Update the .repos files to the new versions for the new distro sync.
Re-run the tests, creating issues for any that failed (Test flow per https://github.com/space-ros/space-ros/issues/102#issuecomment-2064019603)
- [ ] Build basic space-ros image following instructions in https://github.com/space-ros/docker/tree/main/spaceros.
- [ ] Boot into image, run basic commands such as ros2, ikos, just to check that they are installed.
- [ ] Build moveit2 image following instructions in https://github.com/space-ros/docker/tree/main/moveit2.
- [ ] Build navigation2 image following instructions in https://github.com/space-ros/docker/tree/main/navigation2.
- [ ] Build space_robots image following instructions in https://github.com/space-ros/docker/tree/main/space_robots.
- [ ] Boot space_robots image. Start Mars rover demo, check that the demo launches, and it's possible to send commands to the rover.
- [ ] Start canadarm demo, check that the demo launches, and it's possible to send commands to the robotic arm following instructions in https://github.com/space-ros/demos/tree/main/canadarm
- [ ] Boot the Nav2 image and run following the instructions in https://github.com/space-ros/docker/tree/main/nav2_demo#run-the-mars-rover-demo-with-nav2-and-slam-toolbox. Check that the robot is localizing.
- [ ] Docker images are already built but only tagged latest, must tag release as release name.

Finish up docs and start next release window
- [ ] Add announcements to Website, Docs, and Discourse
- [ ] Create next release milestone and tracking issue

After release window

- [ ] Open issues based on release experience
- [ ] Document lessons learned

eholum, 2024-07-31:
- [X] Release tag: `humble-2024.07.0`
- [X] Release issue: #155 (self referential)
- [ ] Milestones
  - [X] https://github.com/space-ros/space-ros/milestone/7
  - [X] https://github.com/space-ros/docker/milestone/3
  - [X] https://github.com/space-ros/docs/milestone/3
  - [X] https://github.com/space-ros/demos/milestone/3
  - [ ] https://github.com/space-ros/requirements/milestones
  - [ ] https://github.com/space-ros/process_sarif/milestones
  - [ ] https://github.com/space-ros/simulation/milestones
  - [ ] https://github.com/space-ros/spaceros.org/milestones
  - [ ] https://github.com/space-ros/dashboard/milestones
- [X] Release project view: https://github.com/orgs/space-ros/projects/3/views/9

eholum, 2024-07-31:
@Bckempa Here's the updated checkboxes based on my/your testing. And hopefully a complete list of the remaining buttons to push:

- [X] Built moveit2 image
- [X] Built navigation2 image
- [X] Built space_robots image
- [X] Built nav2_demo image
- [X] Mars rover test
  - On my arm64 VM (mac w/parallels) I have to run with software acceleration: `LIBGL_ALWAYS_SOFTWARE=1`, it's crazy but it runs and works
- [X] Canadarm test
  - Same deal as above 
- [ ] Run the nav2 SLAM demo test
  - Still building this...

In terms of tags that should be made, I think this is the complete list with a brief note about changes:

- [ ] Repos that will need the `humble-2024.07.0` release tag,
  - [ ] https://github.com/space-ros/space-ros - once https://github.com/space-ros/space-ros/pull/184 is done and merged
  - [ ] https://github.com/space-ros/demos - No changes since the last release
  - [ ] https://github.com/space-ros/docker - I believe `main` is up to date
  - [ ] https://github.com/space-ros/spaceros.org - No changes since the last release
  - [ ] https://github.com/space-ros/process_sarif - No changes since the last release
  - [ ] https://github.com/space-ros/simulation - No changes since the last release
  - [ ] https://github.com/space-ros/docs - No changes from the last release, but Ezra did tag it April