space-ros/space-ros #157: Support Jazzy Jalisco

mkhansenbot, 2024-05-16:
ROS 2 Jazzy will be released by the end of May. We need to discuss what we want to do, I think the options are:
1. Ignore it for this release (2024.7.0) - release humble only
2. Move main to be based on jazzy and release only jazzy going forward
3. Branch humble to it's own branch, move main to jazzy and release jazzy, continuing to update / release humble when there's a new sync release, but using main (jazzy) for new development

EzraBrooks, 2024-05-16:
Agreed with all of @Bckempa's thoughts here.

EzraBrooks, 2024-05-30:
We should have switching rosdistros as an acceptance criteria for the Docker refactor work

asimonov, 2024-08-12:
> I'm willing to champion this for the next release, but I know @asimonov wants to advocate for sticking with Humble for a bit longer. Thoughts?

now I understand the internals of space-ros project better I am actually split. 

on one hand it would be great to move to jazzy to enable new functionality and better versions of nav2, gazebo etc. it will also probably make supporting space-ros easier.

on the other hand it will probably require significant effort to migrate all of the demos and verify everything still works. 

another concern is support on less mainstream embedded platforms. jazzy requires moving to ubuntu24. i would not expect most of those platforms to support ubuntu24 for 1-2 years. arm64 may be less of a problem, but risc-v is. 

how about we try to create jazzy branch and try to make it work on amd64 first? if the process converges by time of next release then we can discuss making jazzy branch main and discuss supporting humble branch for some time?

EzraBrooks, 2024-08-14:
[Ubuntu 24.04 includes mainline RISC-V support](https://wiki.ubuntu.com/RISC-V), but I'm not sure about the completeness of its Apt repositories.

asimonov, 2024-08-15:
> [Ubuntu 24.04 includes mainline RISC-V support](https://wiki.ubuntu.com/RISC-V), but I'm not sure about the completeness of its Apt repositories.

interesting. I have Microchip ICICLE kit, so can try to install ubuntu that they provide. but i did not even know there is ubuntu 22 for ICICLE. The standard SDK is using yocto. so I do not hold my breath for ubuntu24 working out of the box. and yes, risc-v apt repos probably miss a lot of ROS requirements.

lets start with amd64 source build to see the damage

asimonov, 2024-08-22:
> > How can we check? what would upgrading to jazzy actually involve for `space-ros`? switch to ros2.repos from jazzy?
> 
> We set the ubuntu and ros version variables in the Earthly file and build the main space-ros image. That should "just work"™️ (and if it doesn't, that's a bug) but then it will get interesting when we move into the demos.
> 

Oh, I did not realise it is that easy. Out of interest I just tried building space-ros using ubuntu24 and ROS_DISTRO=jazzy.

it fails on ikos (rosdep stage):

```
             +rosdep | In file included from /opt/spaceros/ikos/core/include/ikos/core/adt/patricia_tree/utils.hpp:48,
             +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/adt/patricia_tree/set.hpp:57,
             +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/linear_constraint.hpp:51,
             +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/value/numeric/interval.hpp:55,
             +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/value/machine_int/interval.hpp:50,
             +rosdep |                  from /opt/spaceros/ikos/ar/include/ikos/ar/pass/simplify_upcast_comparison.hpp:49,
             +rosdep |                  from /opt/spaceros/ikos/ar/src/pass/simplify_upcast_comparison.cpp:44:
             +rosdep | /opt/spaceros/ikos/core/include/ikos/core/semantic/indexable.hpp:56:20: error: ‘uintptr_t’ in namespace ‘std’ does not name a type
             +rosdep |    56 | using Index = std::uintptr_t;
             +rosdep |       |                    ^~~~~~~~~
             +rosdep | /opt/spaceros/ikos/core/include/ikos/core/semantic/indexable.hpp:1:1: note: ‘std::uintptr_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
```

I did not look into details, but I presume it is due to compilers versions being different.

asimonov, 2024-08-22:
BTW, I had to move ROS_DISTRO ENV to the top of Earthfile (https://github.com/space-ros/space-ros/blob/main/Earthfile#L27), otherwise rosdep has it empty somehow when we start FROM ubuntu:noble. 

somehow it works ok in current jammy build, which i think may be a hidden bug...

xfiderek, 2024-08-22:
We need to scan through https://github.com/space-ros/space-ros/issues/107 and see which of the demos use .repos file or have hardcoded rosdistros. we will need to fix them

asimonov, 2024-08-22:
> > > How can we check? what would upgrading to jazzy actually involve for `space-ros`? switch to ros2.repos from jazzy?
> > 
> > 
> > We set the ubuntu and ros version variables in the Earthly file and build the main space-ros image. That should "just work"™️ (and if it doesn't, that's a bug) but then it will get interesting when we move into the demos.
> 
> Oh, I did not realise it is that easy. Out of interest I just tried building space-ros using ubuntu24 and ROS_DISTRO=jazzy.
> 
> it fails on ikos (rosdep stage):
> 
> ```
>              +rosdep | In file included from /opt/spaceros/ikos/core/include/ikos/core/adt/patricia_tree/utils.hpp:48,
>              +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/adt/patricia_tree/set.hpp:57,
>              +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/linear_constraint.hpp:51,
>              +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/value/numeric/interval.hpp:55,
>              +rosdep |                  from /opt/spaceros/ikos/core/include/ikos/core/value/machine_int/interval.hpp:50,
>              +rosdep |                  from /opt/spaceros/ikos/ar/include/ikos/ar/pass/simplify_upcast_comparison.hpp:49,
>              +rosdep |                  from /opt/spaceros/ikos/ar/src/pass/simplify_upcast_comparison.cpp:44:
>              +rosdep | /opt/spaceros/ikos/core/include/ikos/core/semantic/indexable.hpp:56:20: error: ‘uintptr_t’ in namespace ‘std’ does not name a type
>              +rosdep |    56 | using Index = std::uintptr_t;
>              +rosdep |       |                    ^~~~~~~~~
>              +rosdep | /opt/spaceros/ikos/core/include/ikos/core/semantic/indexable.hpp:1:1: note: ‘std::uintptr_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
> ```
> 
> I did not look into details, but I presume it is due to compilers versions being different. (there are more compile errors, this is just first one)

may be same issue as https://github.com/NASA-SW-VnV/ikos/issues/273

but not sure why it works in ubuntu22 Earthfile...

asimonov, 2024-08-24:
Houston, we may have a problem...

I keep trying to build Jazzy/Noble combination after manual fix to ikos.

here are the things I found to fix various errors at different times:

```
sudo apt install libacl1-dev -y
sudo apt install libtinyxml2-dev
sudo apt install libeigen3-dev
sudo apt install python3-lark
sudo apt install python3-numpy
```

I also have to skip *cobra_vendor* because not sure how to fix it:

```
cobra_lib.c: In function ‘load_map’:
cobra_lib.c:1451:52: error: ‘%s’ directive output may be truncated writing up to 1021 bytes into a region of size between 2 and 1023 [-Werror=format-truncation=]
 1451 |                 {       snprintf(a, sizeof(a), "%s/%s", C_BASE, s);
      |                                                    ^~
In file included from /usr/include/stdio.h:980,
                 from cobra_fe.h:12,
                 from cobra.h:10,
                 from cobra_lib.c:7:
In function ‘snprintf’,
    inlined from ‘load_map’ at cobra_lib.c:1451:5:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:54:10: note: ‘__builtin___snprintf_chk’ output between 2 and 2044 bytes into a destination of size 1024
   54 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   55 |                                    __glibc_objsize (__s), __fmt,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   56 |                                    __va_arg_pack ());
      |                                    ~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [<builtin>: cobra_lib.o] Error 1
gmake[2]: *** [CMakeFiles/cobra-4.1.dir/build.make:86: cobra-4.1/src/cobra-4.1-stamp/cobra-4.1-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/cobra-4.1.dir/all] Error 2
gmake: *** [Makefile:134: all] Error 2
```

I also have to skip *python_orocos_kdl_vendor* because of:

```
In file included from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/cast.h:16,
                 from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/attr.h:13,
                 from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/pybind11.h:13,
                 from /opt/spaceros/build/python_orocos_kdl_vendor/_deps/python_orocos_kdl-src/python_orocos_kdl/PyKDL/PyKDL.h:26,
                 from /opt/spaceros/build/python_orocos_kdl_vendor/_deps/python_orocos_kdl-src/python_orocos_kdl/PyKDL/PyKDL.cpp:27:
/opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/detail/type_caster_base.h: In function ‘std::string pybind11::detail::error_string()’:
/opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/detail/type_caster_base.h:482:26: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  482 |             frame = frame->f_back;
      |                          ^~
In file included from /usr/include/python3.12/Python.h:42,
                 from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/detail/common.h:215,
                 from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/pytypes.h:12,
                 from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/cast.h:13:
/usr/include/python3.12/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
Summary: 108 packages finished [54.8s]
  1 package failed: python_orocos_kdl_vendor
```

but then this bind11 issue happens in *rclpy*, at which point I am not sure what to do:

```
...
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:244: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/graph.cpp.o] Error 1
In file included from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/cast.h:16,
[... 9 lines cut ...]
                 from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/pytypes.h:12,
                 from /opt/spaceros/install/pybind11_vendor/include/pybind11_vendor/pybind11/cast.h:13:
/usr/include/python3.12/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:258: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/guard_condition.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:132: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/action_server.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:202: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/duration.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:188: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/destroyable.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:90: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/_rclpy_pybind11.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:216: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/clock_event.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:286: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/logging.cpp.o] Error 1
gmake[2]: *** [CMakeFiles/_rclpy_pybind11.dir/build.make:174: CMakeFiles/_rclpy_pybind11.dir/src/rclpy/context.cpp.o] Error 1
[... cut ...]
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rclpy [37.3s, exited with code 2]
Aborted  <<< rclcpp [2min 10s]                                            
```

What I do not understand is how it works with standard ROS2 Jazzy build from source, as recommended (https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Development-Setup.html). I believe we are doing everything same way...

Anyone wants to reproduce or has ideas how to move forward?

Mechazo11, 2024-08-24:
@asimonov thanks for reporting the ```cobra_vendor``` issue. I was very curious to see how to build ```space_ros``` natively in my local machine and also found out I could no longer install ```cobra_vendor```. So every time I now rebuild the workspace, I have to manually ignore it using ```--packages-ignore cobra_vendor```. Would you kindly open a separate Issue for ```cobra_vendor```?

Regarding ```ikos``` I am waiting for confirmation from Dr. Ivan. If he says I can add the other two files without needing to sign another CLA, I will do so post-haste.

asimonov, 2024-08-28:
> @asimonov thanks for reporting the `cobra_vendor` issue. I was very curious to see how to build `space_ros` natively in my local machine and also found out I could no longer install `cobra_vendor`. [...]

I am not actually sure what cobra_vendor does. But it works as part of our current Earthfile (ubuntu22). Let me investigate first so i can describe how to reproduce the issue...

ivanperez-keera, 2024-09-05:
Should the issue with Cobra be reported to the cobra repo or is this an issue specifically with Space ROS?

asimonov, 2024-09-08:
> Should the issue with Cobra be reported to the cobra repo or is this an issue specifically with Space ROS?

we can do. but may be we can try upgrading cobra version in ament_cobra? at the moment it references 4.1 (https://github.com/ament/ament_cobra/blob/master/cobra_vendor/CMakeLists.txt#L13). Looks like current cobra version is 4.7. I just tried to change ament_cobra locally to use 4.7, but colcon still fails. But looks like different error, during install phase

ivanperez-keera, 2024-09-09:
I'm not sure what's going on.

The bin linux offered by cobra is a directory. See: https://github.com/nimble-code/Cobra/releases/tag/version-4.7 (which offers a .zip that inside has a directory).

Inside that there should be a `cobra` binary. I tried it on my machine and works as expected.

Yes, there's a tcl script there somewhere in the same zip file, but that should not be what you are running.

asimonov, 2024-09-09:
> I'm not sure what's going on.
> 
> The `bin_linux` offered by cobra is a directory. See: https://github.com/nimble-code/Cobra/releases/tag/version-4.7 (which offers a .zip that inside has a directory with several executables).
> 
> Inside that there should be a `cobra` binary. It's an ELF. I tried it on my machine and it works as expected.
> 
> Yes, there's a tcl script there somewhere in the same zip file, but that should not be what you are running.

i think I need to understand how ament_cobra builds then...

asimonov, 2024-09-09:
> I'm not sure what's going on.
>
> The `bin_linux` offered by cobra is a directory. See: https://github.com/nimble-code/Cobra/releases/tag/version-4.7 (which offers a .zip that inside has a directory [...]

@ivanperez-keera ok, looks like bin_linux directory is missing in ubuntu24 clean checkout setup.. there is a PR in cobra repo, I commented on PR fixing this issue: https://github.com/space-ros/space-ros/pull/205#issuecomment-2337440887. shall we continue these discussions about fixing particular dependency issues on PR?

But in general we'll need that cobra PR to be merged and new release (4.8) made so we can use it in our Jazzy upgrade, similar to ikos plan. Do you know anyone supporting cobra to help us? That PR was there since March

asimonov, 2024-09-19:
update on the current status:
* cobra fix merged (https://github.com/nimble-code/Cobra/pull/68). need a new release
* ikos fix is still open (https://github.com/NASA-SW-VnV/ikos/pull/274). waiting for merge and the new ikos release
* question is still open if we need to move `space-ros` branch of `ament_lint` forward to match Jazzy: https://github.com/space-ros/space-ros/pull/205#issuecomment-2337998327
* need to add image push from jazzy branch

asimonov, 2024-10-29:
current status on PR: https://github.com/space-ros/space-ros/pull/205#issue-2507656377

basically build and tests run ok.

someone needs to fast forward `jazzy` branch. i have worked on latest state of main, so some commits from outside my PR are shown in the PR. this should be gone once `jazzy` branch is synched.

test results:
* i have compared humble/main and jazzy test output artifacts
* the comparison was not using VScode Sarif plugin, but comparing the output directories
* in humble/main we generate output for 103 packages, in jazzy for 74. mostly due to cmake_xxx packages sarif missing. but in general jazzy has slightly different package list
* `copyright`, `clang_tidy` sarif output is missing on jazzy for some reason 
* cobra sarif is same, but the format is slightly different due to new version of cobra

asimonov, 2024-10-30:
i have rewritten commit history on this branch for clarity.
it all builds and runs as before.

what is left:
* please sync up jazzy branch to main
* please kick off the CI
* once we have ament_cobra tag i'll update spacros.repos and ros2.repos 
* we need to agree if we push jazzy branch to dockerhub (like main) only or we do jazzy release too

Mechazo11, 2024-12-02:
Hi @asimonov I was wondering, since Space ROS is moving to ROS 2 Jazzy, is Gazebo Harmonic now going to become the default Gazebo engine? I think in the last release, the officially supported version was Gazebo Fortress (Ignition) if I am not mistaken. As part of a different project, I was able to successfully integrate Gazebo Harmonic with a Jazzy workspace both built from source in Ubuntu 22.04. I will be happy to assist integrating Harmonic with Space ROS for the jazzy version.

asimonov, 2024-12-03:
> Hi @asimonov I was wondering, since Space ROS is moving to ROS 2 Jazzy, is Gazebo Harmonic now going to become the default Gazebo engine? I think in the last release, the officially supported [...]

hi @Mechazo11, 

yes, I think we'll move to Jazzy+Harmonic imminently. it would be great if you help with brining demos up to speed on Harmonic!

PS: sorry, I never had chance to reproduce your docker build error. was on my list for ages, but it was very busy times

asimonov, 2024-12-03:
@Bckempa @EzraBrooks 

shall we merge jazzy to main then? do we need a ticket for this?

Mechazo11, 2024-12-03:
Hi @asimonov no worries, I too have stopped working with O3DE for the time being. As advantageous as it may seem initially, the project is still in its infancy and would take a significant time, effort (and money) to bring it up to the maturity as Gazebo is in the robotics community. I will go ahead and close that ticket for the time being.

On a different note, @Bckempa, @EzraBrooks I was wondering, is the package roster for jazzy branch finalized yet? I recall seeing in this Issue #196 that some discussion took place regarding the pros and cons of adding ros2_control into the base image.  I was wondering if the team has reached a consensus regarding this matter? From my recent experience working with Gazebo Harmonic I think it is strategically advantages to have ```ros2_control``` and its dependency packages in the base image.

commit d792a219 (2024-11-22): Merge pull request #205 from asimonov/157_jazzy_upgrade

upgrade space-ros to jazzy/noble (closes #157)