space-ros/space-ros #158: Adjust main spaceros image to the new build approach

xfiderek, 2024-05-28:
See discussion https://github.com/space-ros/space-ros/discussions/140 for more context

TODOs:
- Move Earthfile from `docker` to `spaceros` repo
- Swap current `generate_repos.sh` file with the new version that will make it easier to avoid code duplication between spaceros image and downstream ones.  (see below)
- Include `generate_repos.sh` in the final image (e.g. /usr/bin)
- Rework current Earthfile to make it possible to save base and dev images, as per discussion
- To figure out - final naming convs or base and dev images
- Decide if we want to push base image to docker hub
- Document both images in README

The proposed modified version of generate_repos.sh https://gist.github.com/xfiderek/e07e02930d58c05807208dee2742d9a1

xfiderek, 2024-05-30:
@bckempa suggestion - reconsider path / naming to the generate_repos.sh

eholum, 2024-05-31:
I don't seem to have the power to assign myself issues but I'll plan on taking this on starting next week. @xfiderek can you assign to me?

xfiderek, 2024-06-01:
I don't have permissions either. @EzraBrooks @bckempa @mkhansenbot @ivanperez-kerra

eholum, 2024-06-09:
Step 1 to move the Earthfile and related content from `docker` to here is being done in : https://github.com/space-ros/space-ros/pull/163.

As noted in the PR, I think it's preferable to test/merge that before moving onto other tasks that would make further modifications to the build process.

eholum, 2024-06-23:
Still waiting on reviews for the two open PRs, but in the interim @xfiderek here is the basic consolidation of the repos file generation with the main build that you mentioned if you wanted to take a look. I just merged the two open PRs and made the changes based on that. It's WIP but give you an idea of what I was talking about...

https://github.com/eholum/space-ros/pull/1

eholum, 2024-07-26:
This will be resolved in https://github.com/space-ros/docker/pull/151.

asimonov, 2024-07-29:
should this one be moved to `docker`? 

in `space-ros` we have #172 to cover this work?

eholum, 2024-07-29:
Can we change this issue's repo? That would make sense to me as the PR is in the `docker` repo. We could also close this and open an alternate issue... As long as everything is linked from https://github.com/space-ros/space-ros/issues/177.