space-ros/space-ros #159: Create lightweight, core image for spaceros

xfiderek, 2024-05-28:
See discussion https://github.com/space-ros/space-ros/discussions/140 and issue https://github.com/space-ros/space-ros/issues/107 for more context
TODOs
- Add a stripped down version of spaceros image that inherits from spaceros dev, with removed bloat, that contains minimal binaries required to run spaceros packages and build on top of it.
- Document the image in README

franklinselva, 2024-08-08:
I did an exercise of striping the `core` and `core` images of spaceros and ended up bringing the images down to 1GB and 1.3GB respectively. Is anyone working on this ticket? If not, I would be happy to take it out to my plate.

franklinselva, 2024-08-08:
I am glad to hear.

I came across this PR and there is some overlap already. I will have to wait until a decision is made on #170. Can you provide some context on source clone caching? I recently went through the docs of `earthly` and something similar hit.

franklinselva, 2024-08-16:
> > Can you provide some context on source clone caching? I recently went through the docs of `earthly` and something similar hit.
> 
> The discussion was around where to do the vcs-tool clone of the base ROS packages to keep them cached without inflating the image size, both for CI and for local builds. We have a few options. @xfiderek and @EzraBrooks have more thoughts on this and are going to open an issue/discussion in the near future to work that out.

I think we can use earthly's [`GIT CLONE`](https://docs.earthly.dev/docs/guides/best-practices#git-clone-vs-run-git-clone) and [`FOR`](https://docs.earthly.dev/docs/guides/best-practices#for-...-in-...-vs-run-for-...-in)  syntax for it. The only addition here would be convert the VCS repos file to a list of git URLs with commit IDS.

EzraBrooks, 2024-08-19:
If we're going to change the way we track Git dependencies, I think we should opt to use Git submodules to make it as tooling-agnostic as possible.

I know this is a bit of a controversial opinion in the ROS ecosystem, but I have _always_ had better luck shipping submodules than shipping vcstool dependencies.

franklinselva, 2024-08-19:
That's an interesting alternative approach to `vcs`.

`vcs` is a tool for modules which often see frequent updates with handful of git operations to work with. We are not using this tool to the full extent. 

On the other hand, I don't see the need of frequent updates of the modules where git submodules shine. But I think it would add more complexity to consider for entire source build.

EzraBrooks, 2024-08-19:
My point was more that:
> The only addition here would be convert the VCS repos file to a list of git URLs with commit IDS.

This is basically just using submodules but with extra steps. Submodules are just a Git URL and a reference to a commit hash :)

franklinselva, 2024-08-19:
I get it and I still see it as a complexity in managing it inside the project directly. The difference here would be submodules needs manual work to maintain it and `vcs` and the conversion script automate the process.

EzraBrooks, 2024-08-20:
> The difference here would be submodules needs manual work to maintain it and vcs and the conversion script automate the process.

I'm not sure I understand what you're saying here - Space ROS intends to fix its dependency versions, whereas the main advantage of `vcs` is targeting rolling HEADs.

asimonov, 2024-09-04:
> > The difference here would be submodules needs manual work to maintain it and vcs and the conversion script automate the process.
> 
> I'm not sure I understand what you're saying here - Space ROS intends to pin its dependency versions, whereas the main advantage of `vcs` is targeting rolling HEADs.
> 
> However, I also acknowledge that the ROS community is allergic to submodules, so I won't push too hard on this topic.

oh no, not submodules! 

they seem like a nice idea, but they never work with a project with more than 1 developer IMHO.

EzraBrooks, 2024-09-04:
As I said, I won't press for them, because apparently everyone in the robotics community holds that same belief.

FWIW we have like fifteen submodules in MoveIt Pro on a team of >10 engineers and it works fine once everyone learned how submodules work.

No more submodules-related conversation here since it's off-topic :)

asimonov, 2024-09-19:
> As I said, I won't press for them, because apparently everyone in the robotics community holds that same belief.
> 
> FWIW we have like fifteen submodules in MoveIt Pro on a team of >10 engineers and it works fine once everyone learned how submodules work. It's interesting to me how people accept that "Git is hard, but extremely powerful when you learn how to use it" but then don't extend that same axiom to submodules.
> 
> No more submodules-related conversation here since it's off-topic :)

I think submodules merge two things together - the management of reference repo information (like our repos files) and the git machinery of clone/branch/merge/commit. What it results in is every git operation starts to have non-trivial side effects, potentially. I wonder what your developer guide looks like in relation to these operations with submodules... And on top of that it does not really resolve the issues of un-synchronised updates to multiple repos.