space-ros/space-ros #171: Build and push arm64 Docker images

EzraBrooks, 2024-07-25:
This will make it easier to develop on Apple Silicon Macs and on NVIDIA Jetson edge compute devices.

EzraBrooks, 2024-07-25:
moved!

dsobek, 2024-09-04:
I'm taking a stab at this issue since I'm on arm64. Earthly has some documentation on how to do multi platform builds:
https://docs.earthly.dev/docs/guides/multi-platform

EzraBrooks, 2024-09-04:
GitHub just released arm64 CI runners to general availability yesterday (?), so great timing.

asimonov, 2024-09-05:
> documentation

arm64 builds on macos docker. but i was wondering if we should go for proper multi-arch builds or have separate image names for different architectures... i do not have experience with multi-arch builds, and it looks like it is more complicated process than what we have right now

asimonov, 2024-09-05:
> Let me know how that goes, @dsobek. I hit a bit of a wall trying to use it for `riscV` builds recently but there were several agrevataing circumstances that made the situation harder than it should be for this.
> 

I have not raised a separate issue for RISC-V build yet. may be it should be a discussion as it is less mature platform and there are multiple ways we can go with it. In my case I have hardware to verify, but have no time yet to start working on it.

EzraBrooks, 2024-09-05:
Thanks for the offer of compute resources @Bckempa, but we should be good on our end - we have a few arm64 MacBooks and also a very burly 64-core Ampere dev kit.

@dsobek if you need the big 64-core one (it builds things extremely fast), let me know on Slack and I can point you in the direction of the credentials.

asimonov, 2024-09-19:
can i ask a related question here...

I am trying to move from x86 laptop to do full space-ros based development on macbook (apple silicon). As I understand there are 3 ways to do it:
* docker on macos
* VM in macos
* dual boot into something like Asahi-based ubuntu

in my experience docker builds of spaceros and other 'ubuntu'-valid images work fine using arm64 architecture. with amd64 I have seen some issues, but this is fine, I would like to only use arm64. but the problem with docker is running GUI - neither Gazebo nor rviz2 work from docker. or at least I could not make it work...

I did not try dual boot solution, because one benefit from using macos is that I can do development and other things on single machine/screen.

VM can be done 3 ways:
* UTM using QEMU. slow as it is emulation
* UTM using Apple Virtualisation (arm64 ubuntu). see below
* Parallels. but that cost money and is pain on each macos upgrade.

I made UTM with Apple Virtualisation work. it can consume arm64 images built on macos docker. (i did not try building images in VM yet). and, more importantly, inside VM I can run Gazebo and all the GUI tools, because it is essentially full Ubuntu environment. The gazebo performance is quite low - only getting 3-5fps on M1 mac with 8 cores. But it works.

any suggestions how to do it better?

dsobek, 2024-09-19:
@Bckempa have you had any trouble using colima on apple silicon? I've been looking at using colima in the macos runner in this PR and I've been testing it on my M1 laptop, however I'm getting a sigkill during the colcon build stage. Using docker desktop seems to work fine.

EzraBrooks, 2024-09-19:
sigkill during build is almost certainly OOM

asimonov, 2024-09-20:
> For both arm64 and x86 docker on my apple silicon laptop I use [colima](https://github.com/abiosoft/colima) which provides access to the underlying [lima](https://github.com/lima-vm/lima) vm which explicitly supports [cross-arch](https://lima-vm.io/docs/config/multi-arch/) VMs including using [Rosetta on VZ.framework](https://lima-vm.io/docs/config/multi-arch/#fast-mode-2-rosetta-intel-containers-on-arm-vm-on-arm-hostfast-mode-2) which is quite performant.
> 
> So far it's been great for multi-arch development but I haven't tried anything GUI with Space ROS, but I have for [astrobee](https://github.com/nasa/astrobee) and that's been fine.

oh, nice!
I was trying colima couple of weeks back for x86 builds on M1 mac. i could not build our main branch of space-ros for some reason. 

to be honest the error was in cobra_vendor, so I suspect it may be same issues as with jazzy build...

the issues looked like running out of file handles:

```
make[3]: *** write jobserver: Bad file descriptor.  Stop.
make[3]: *** Waiting for unfinished jobs....
make[3]: *** write jobserver: Bad file descriptor.  Stop.
gmake[2]: *** [CMakeFiles/cobra-4.1.dir/build.make:86: cobra-4.1/src/cobra-4.1-stamp/cobra-4.1-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/cobra-4.1.dir/all] Error 2
gmake: *** [Makefile:134: all] Error 2
```

so i dived into configuring lima but without luck.

and also at some point I was seeing OOM/sigkills

@Bckempa 
* did you change colima linux VM parameters in any way?
* can you try to build current `space-ros/main` in colima?

related question: how does everyone sizes VMs? I try to allocate 6 cores and 2*cores Gb of RAM. for normal macos docker it works on arm64 build

asimonov, 2024-09-22:
> > * did you change colima linux VM parameters in any way?
> 
> I bumped the cores, ram, and disk space when creating the context but didn't do anything deeper.
> 
> > * can you try to build current `space-ros/main` in colima?
> 
> I'll try that this weekend, I've been building Dockerfiles that start from a space-robots build I pull from an internal registry so far so I haven't built from scratch.
> 
> > related question: how does everyone sizes VMs? I try to allocate 6 cores and 2*cores Gb of RAM. for normal macos docker it works on arm64 build
> > That's in line with what I've been doing, but the VM I use to build the Space ROS and Space Robots images are massive and run on server hardware and then I use my laptop to develop on top of those. I'll see what happens when I build locally.

yes, would be great if you try space-ros/main. i did a few attempts on macos with different VMs but could not get to success with amd64. it would be nice to have, but if it does not work then one can still build using intel machines.

asimonov, 2025-10-17:
now, a year later, I read that github actions do provide free arm64 runners on public repos.

https://github.blog/changelog/2025-08-07-arm64-hosted-runners-for-public-repositories-are-now-generally-available/

shall we try?

ivanperez-keera, 2026-10-01:
Tagging @dv-picknik who is working on this to see if Github lets me mark this as assigned to him.

closed 2026-10-19 by ivanperez-keera

asimonov, 2026-10-20:
yay!

3 years from open to close. to the day :)