space-ros/space-ros #178: Restructure the repository content at Organization level

franklinselva, 2024-07-30:
I am trying to run the demos from space-ros and realized I need to clone three different repositories to play with the nodes and scripts since each one is coupled with the other. I see why it is detached and coupled in specific areas using `vcs` tool. But the question is why the approach is set up the way it has been.

The trouble of starting this issue somewhere different than the `demos` repository is already hinting at the confusion that will arise.  I recently came across this [comment](https://github.com/space-ros/docker/issues/131#issuecomment-2165609708) and the issue didn't fit in the repository either. Plus, I feel the changes represented in this comment are hard for both the developers and reviewers to work with.

### Scenarios

I want to run the demos using dockerization, meaning I need to clone the `docker` repository and not the `demos` repo. This is misleading since the way I can kick off the demos is by building the docker images from the `docker` repo.

## Feature request

Reorganize the structure so the demos can be run in a much simpler way

#### Feature description
The demos are meant to be a quick start and entry guide for new developers into the space-ros. This means there is less need to isolate the simulation assets from the demo packages. I feel like the simulation assets and the corresponding demo ros packages can be coupled together by repackaging the `simulation` repository to be `rover-description` and `canardarm-description`. The essence of docker can be added to the `demos` repository directly instead of setting up at the organization level.

#### What this feature can offer 

- This change could potentially reduce the number of breaking points when working with the demos and adding new demos.
- Easier to manage existing demos and add new simulators
- Less maintenance and dependency conflicts

#### Implementation considerations

The `demos` repository could be structured something like this.

```bash
spaceros-demos/
├── Makefile
├── canardarm-spaceros
│   ├── canadarm-demo
│   ├── canadarm-descripion
│   └── canadarm-moveit-config
├── docker
└── marsrover-spaceros
    ├── marsrover-demo
    └── marsrover-description

8 directories, 1 file
```

This is going to be a breaking change if the request is accepted and is assigned to the milestone. But this will remove the need for two organization-level repositories `simulation` and `docker` and easier management of changes in the longer run.

franklinselva, 2024-07-30:
@Bckempa Considering the issues and discussions mentioned, my proposal would be unaffected for most part. This proposal can be worked comfortably on considering a solution found for building moveit2 and nav2-related layers for the docker image. I see the `osrf/space-ros:latest` already in [dockerhub](https://hub.docker.com/r/osrf/space-ros).

Will I be able to join the next technical committee meeting by any chance?

franklinselva, 2024-07-30:
@Bckempa I think there is no dependency either way. But moving the docker files to the demo repository will impose additional work before improving the organization (if docker files are expected to be merged to main). Reorganizing the higher level items first will help less changes to the docker files since the moveit2 and nav2 will be already staged and ready for demos package.

franklinselva, 2024-07-31:
@Bckempa Adding more context from the [discussion on docker](https://github.com/space-ros/space-ros/discussions/140#discussioncomment-10197621). The image tags `moveit` and `nav2` can be directly used for both the demo applications. This leaves the demo packages to be restructured independently.

quarkytale, 2024-08-01:
> why `demos` and `simulation` were split initially?

Mostly because earlier there were plans to add the same models to play with in different demo worlds. Since the goal of demos is changing, we could reorganize.

mkhansenbot, 2024-08-08:
I think each demo should have it's own `Dockerfile` or `docker-compose.yaml` file so each can be built independently. Moving the `Dockerfile` from the `docker` repo to here and splitting it into two separate files would be a logical first step to doing that which wouldn't require much effort.

franklinselva, 2024-08-08:
Agreed. Something like this for the docker entry points is expected
```Dockerfile
# Canardarm demo
FROM osrf/space-ros:moveit2
...
```
and
```Dockerfile
# Rover demo
FROM osrf/space-ros:nav2
...
```
I don't see if these two demos can be shared in the same environment. But using `Makefile` or `docker-compose` would help manage the docker images easily. I would opt for `Makefile` personally since we can set up a custom stream of sub-commands to help with demos.

franklinselva, 2024-08-11:
> Can you share some examples of what you'd like to use Makefiles for? I'm always hesitant to add a runtime dependency without a strong use-case.

I could think of a few usecases at the moment. 

I want to mention that the proposal is structuring the demos repository as a mono repo. With that said, we should have multiple examples and demos for each robot (e.g. rover can have demos related to perception, navigation, exploration, mission planning, etc.) and to keep things modular, it is better to structure each section of demos in a separate ROS package.

Now, to build the demos, If I am concentrating only on perception for Rover, I am not required to build the mission planning and nav2-related examples of Rover or any other robots in general. This will save a lot of build time when the repository grows.

Something like this would be useful at this point,

```bash
make build curiosity_rover nav2_example # Nav2 related demo
make build curiosity_rover all # All demos of the rover
```

Even though it is not meant to be, I sometimes take the `make` command further for running the commands without tools like [`just`](https://github.com/casey/just).

asimonov, 2024-08-12:
> > Can you share some examples of what you'd like to use Makefiles for? I'm always hesitant to add a runtime dependency without a strong use-case.
> 
> I could think of a few usecases at the moment.
> 
> I want to mention that the proposal is structuring the demos repository as a mono repo. With that said, we should have multiple examples and demos for each robot (e.g. rover can have demos related to perception, navigation, exploration, mission planning, etc.) and to keep things modular, it is better to structure each section of demos in a separate ROS package.
> 
> Now, to build the demos, If I am concentrating only on perception for Rover, I am not required to build the mission planning and nav2-related examples of Rover or any other robots in general. This will save a lot of build time when the repository grows.
> 
> Something like this would be useful at this point,
> 
> ```shell
> make build curiosity_rover nav2_example # Nav2 related demo
> make build curiosity_rover all # All demos of the rover
> ```
> 
> Even though it is not meant to be, I sometimes take the `make` command further for running the commands without tools like [`just`](https://github.com/casey/just).

why dont we use Earthfiles for something like this? they mix makefile functionality with docker functionality in a good way.

franklinselva, 2024-08-12:
My thought is Earthfiles in this case will impose additional complexity in configuring different demos. To address this, the question we want to answer is how the demos should be built and run from here? Should we still have all the demos prebuilt inside image?

My idea here is to build a base docker image for a robot or a family of robots and use makefiles to build demos dynamically within the containers instead of building directly into images.

franklinselva, 2024-08-13:
I think we have considerable acceptance now. Should I start working on this change?

franklinselva, 2024-08-13:
Having worked with the build system closely, I think most of the issues listed in #107 don't need to be carried down. #177 is already handled by #186 to an extent which needs to be split down to stick with the guidelines and NASA standards (I will still need some help on this).

To expand on the situation for #107, I have created a new issue #188 which should help determine the changes needed before proceeding.

This is what I would propose to be done in order
 - #186 & #177
 - Discuss #188
 - Decide the changes for nav2 and moveit2
 - #107 
 - Integrate docker into the demo workspace

This leaves the other changes of the demo to be performed in parallel.

I believe the above list is essential and has ended up as a part of restructuring the organization's content.

franklinselva, 2024-09-09:
Sure. I will do that. We can also move this issue if it is easier.

closed 2025-03-27 by mkhansenbot