space-ros/space-ros #185: Please add a LICENSE file

gmacario, 2024-08-07:
See https://github.com/ros2/ros2/issues/1590

asimonov, 2024-08-08:
interesting point. 

for ros2 repo's issue the comment is that ros2 is meta-information repository, there is no code in it, so there is nothing for license to cover.

for space-ros repo we have some docker/build/CI code, so may be good idea to add a license

gmacario, 2024-08-26:
> for space-ros repo we have some docker/build/CI code, so may be good idea to add a license

Based on https://github.com/ros2/ros2/issues/1590#issuecomment-2307499711 I concur.

gmacario, 2024-09-03:
Hello space-ros maintainers,

> > for space-ros repo we have some docker/build/CI code, so may be good idea to add a license
> 
> Based on [ros2#1590 (comment)](https://github.com/ros2/ros2/issues/1590#issuecomment-2307499711) I concur.

Shall @mkhansenbot comment at https://github.com/space-ros/demos/issues/29#issuecomment-2274125338 apply to this repository as well?