space-ros/space-ros #187: Unable to build ```space-ros```` using official instructions

Mechazo11, 2024-08-10:
- Operating System:
  - Ubuntu 22.04
- Installation type:
  - First time build with Earthly
- Version or commit hash:
  - Not sure, assuming ```humble-2024.07.0```
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Step 2 **2 Bootstrap the Earthly configuration file.** from the official [first time build](https://space-ros.github.io/docs/rolling/Tutorials/First-Time-Building.html) tutorial. 

Command: ```earthly --artifact github.com/space-ros/docker/spaceros:nuclearsandwich/earthly-prototyping+earthfile/Earthfile```

#### Expected behavior
Setup dependencies and create ```src``` with ```space-ros``` core packages. Expected to use the latest build of ```space-ros```

#### Actual behavior
No files are created and an error message attached below is shown. Unable to proceed with building ```space-ros```

```
Error: resolve build context for target github.com/space-ros/docker/spaceros:nuclearsandwich/earthly-prototyping+earthfile: 
failed resolving git project [platform: linux/amd64]: read uname-m: failed to load cache key: repository does not contain
ref nuclearsandwich/earthly-prototyping, output: ""
```

Requesting help in solving this issue.

syedjawadakhtar, 2024-08-11:
Hi @Mechazo11,

Have you tried following [these](https://github.com/space-ros/space-ros/blob/main/docs/USAGE.md) instructions for the building the `space-ros` docker image?

Mechazo11, 2024-08-11:
Hi @Bckempa thanks for the heads up. Yes , I was able to get space-ros built using instructions from the [Usage.md](https://github.com/space-ros/space-ros/blob/main/docs/USAGE.md).  

```
cd ~
mkdir space_ros && cd space_ros
git clone https://github.com/space-ros/space-ros.git
cd space_ros
./build.sh # Build osrf/space-ros image
./run.sh # Test osrf/space-ros image
exit # Exit out of the docker container
```

closed 2024-08-11 by Mechazo11