space-ros/space-ros #196: Control packages for SpaceROS

franklinselva, 2024-08-21:
The current ROS2 stacks available in `space-ros/docker` are `nav2` and `moveit2`. The `ros2_control`  and the robot control in general are missing in the source build packaging and I think it is a must-have stack considering the robot demos that are currently in place.

I would propose to build a slim version of the control stack for the robots to be used with space ROS.
 - Build a slim version of the control stack for Space ROS
 - The kinematics and control algorithms of the Rover and the Canadarm are something that can be managed in a repository of its own.
 - The control algorithms can be written in CPP to demonstrate how spaceROS development of new packages is enforced.

This should improve the complexity of the demos we are aiming for.

Mechazo11, 2024-09-03:
I think when building the ```space_robots``` image, ```gz_ros2_control``` is installed which if I am not mistaken performs the velocity and joint position control for the rover. But [demo_manual_pkgs.repos](https://github.com/space-ros/docker/blob/main/space_robots/demo_manual_pkgs.repos) that makes up the workspace does not have ```ros2_control```. @franklinselva is ```gz_ros2_control``` a stand-alone derivative of ```ros2_control``` or dependent on the latter cuz I did not see ```ros2_control``` being installed into the workspace based on this [repos)(https://github.com/ros-controls/gz_ros2_control/blob/rolling/doc/index.rst) file

asimonov, 2024-09-04:
good idea!

as i understand ros2_control is the controllers package for ROS2 and gz_ros2_control is the Gazebo plugins to support ros2_control. I would say we need both as most of our demos are in Gazebo.

one observation: in my experiments to upgrade to Gazebo Harmonic gz_ros2_control brings gz-sim and gz-plugins, and lots of gz_..._vendor packages which in turn bring some x11 libs etc. so this is only for dev/simulation image, i think...

just to clarify - you want to add ros2_control to space-ros main image? or shall we keep them in demos/stacks?

Mechazo11, 2024-09-04:
@asimonov I am still a novice when it comes to ros2_control so I don't have enough experience with it to make a comment. But having said that I am inclined to agree with @franklinselva in having ```ros2_control``` being a native into ```space_ros``` stack. To me, ros2_control is a like a "ROS inside a ROS" whose main goal (as much as I understood it) was to ease bidirectional control between a ```controller``` and a ```physical (or simulated)``` entity (robots, motors, arms etc.) with ros2_control taking care of converting said commands, temporal dependencies etc.

franklinselva, 2024-09-05:
@asimonov @Mechazo11 There is quite a catch in placing where this control stack should go. Essentially `nav2` and `moveit2` will depend on this `control` stack to allow the robots to be controlled. The main `space-ros` image is a `core` image without any additional dependencies making it an absolute base variance.

This flow could be something better and scalable.

```mermaid
graph LR
  A[Core space-ros image] --> B[Control Image Stack]
  B[Control Image Stack] --> C[Navigation2 Image Stack]
  B --> D[Moveit2 Image Stack]
  C --> E[Robot Demo Image]
  D --> E

subgraph Main repository
  A
end

subgraph Stack Repository
  B
  C
  D
end

subgraph Demo Repository
  E
end
```

Mechazo11, 2024-09-05:
@franklinselva I will be happy to help you with this tutorial. We may also incorporate some of the stringent code checks space-ros is looking to enforce. This would be an ideal tutorial for newcomers into the space-ros stack.

franklinselva, 2024-09-07:
@Mechazo11 That would be a great help. I am working on the docker images of space ros and will be able to work on this rover control soon after that. I am already halfway through the [paper](https://www-robotics.jpl.nasa.gov/media/documents/jfr-trctl-rob-21903-2019.pdf) and this should help with the control implementation of the rover. This helps kick-start the setup in the meantime.

Eventually, the idea is that the algorithm should go into its repository with CPP Implementation and Python wrappers around it.

asimonov, 2024-09-12:
I have just gone through some setting up rover example in Gazebo/space-ros. I agree with @franklinselva that ros2_control is usually not required. Gazebo provide some control framework. I think nav2 does not use ros2_control.

Looking at ros2_control it is quite high abstraction framework to manage/swap controllers (using 'controller manager') and same for real hardware drivers (using 'resource manager')

My up-to-date opinion is to not include ros2_control in basic space-ros. 
We can add it to 'stacks' when required, as an example.

asimonov, 2024-09-12:
> @asimonov @Mechazo11 There is quite a catch in placing where this control stack should go. Essentially `nav2` and `moveit2` will depend on this `control` stack to allow the robots to be controlled. [...]

comment on that diagram - i dont think nav2 depends on ros2_control. does moveit2 depend on it?

franklinselva, 2024-09-12:
I didn't describe things clearly. This is not a dependency graph but a data flow between layers of the images.

asimonov, 2024-09-12:
i am more interested in dependency management. (and version management, but that is separate discussions)

from the point of view of dependency managmement there is no real need to have ros2_control in main space-ros, is there? so why not push it to demos/stacks where it is actually used/depended on?

franklinselva, 2024-09-12:
Ideally, the dependency management depends on the demos and applications we are trying to build. What we are trying to offer are the frameworks and packages strictly defined for space applications. In this case, it's better to use Earthly than Docker to build the base image one would want to use.

Let me break down the dependency for different stages of the rover demo as an example.

For the control demo that is currently in place,

```mermaid

graph LR
  A[Base Image]
  B[Control Stack Image]
  C[Demo Image]
  A --> B
  B --> C

subgraph Main
  A
end

subgraph Stacks
 B
end

subgraph Demo
 C
end
```

If we are adding navigation to the demo, naturally this will be the dependency,

```mermaid

graph LR
  A[Base Image]
  B[Control Stack Image Layers]
  D[Nav2 Stack Image Layers]
  C[Demo Image]
  A --> B
  B --> D
  D --> C

subgraph Main
  A
end

subgraph Stacks
 B
 D
end

subgraph Demo
 C
end
```

The above is the same case for `moveit2` as well where `moveit2` sends the trajectory planned and the `ros2-controllers` convert the trajectory to the list of joint states.

For the above dynamic build, using earthly here is advantageous. But this is a part of a different discussion. The above approach also keep the dependencies and versioning manageable if we are setting up on stacks.