space-ros/space-ros #199: Simplify rosdep command in Earthfile

xfiderek, 2024-09-03:
For now the rosdep command in our Earthfile looks like this
https://github.com/space-ros/space-ros/blob/57649e50be93faa494777e70308c95bb19a7338e/Earthfile#L174-L180

Ideally `--skip-keys` should contain only `$(tr '\n' ' ' < 'excluded-pkgs.txt')`.

1. We need to remove `ikos` from `package.xml` definitions, so that we don't need to exclude it in rosdep - We had PR open but I closed it for some reason. Now I think I closed it accidentally - https://github.com/ament/ament_ikos/pull/3
2. Lets 2x check if we still need to skip `urdfdom_headers` - i tried rosdep install as here and did not have to include it: https://github.com/space-ros/docs/issues/34#issuecomment-2325935599

asimonov, 2024-09-04:
urdfdom_headers seem to be coming from original ROS2 source install docs: https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

not sure if it is needed some way down in builds...

ivanperez-keera, 2026-05-11:
@asimonov @xfiderek Is this still an issue? Is this clearly actionable?