space-ros/space-ros #207: Link to Contributing.md is broken

Mechazo11, 2024-09-07:
Hi,

The link to the [CONTRIBUTING.md][https://github.com/space-ros/space-ros/blob/main/CONTRIBUTING.md] link shows a 404 not found error. This link was part of the instructions for submission in the ongoing NASA Space Sim competition.

Requesting a fix for this link so that I can confirm my submission is adhering to the Space ROS contribution protocols.

With best,
@Mechazo11

gmacario, 2024-09-08:
Hello @Mechazo11,

If you are referring to this link in the toplevel [README.md](https://github.com/space-ros/space-ros#) of this repository

> # Contribution rules
>
> See the [contributing guide](https://github.com/space-ros/.github/blob/master/CONTRIBUTING.md) for details on how to contribute to the Space ROS project.

at least for me it works and correctly links to the existing <https://github.com/space-ros/.github/blob/master/CONTRIBUTING.md>

[image]

Mechazo11, 2024-09-08:
@gmacario thank you very much. I will close the issue.

closed 2024-09-08 by Mechazo11