space-ros/space-ros #215: Conflict between ```rmw_cyclone_cpp``` and ```rmw_implementation``` when building projects that rely on rclcpp

Mechazo11, 2024-09-21:
### Version or commit hash

humble-2024.07.0

### RMW Implementation

CycloneDDS

### Steps to reproduce issue

Dear Space ROS maintainers.

Apologies in advance for the long message but replicating this error and its tentative solution, both requires some convoluted process.

As part of my NASA Space ROS Summer Sprint Challenge 2024 (space-ros/demos PR [#64](https://github.com/space-ros/demos/pull/64)) submission, I demonstrated Open 3D Engine (O3DE) can successfully integrate with Space ROS (any ROS 2 variant to be more exact) via the ROS 2 Gem from O3DE project. ROS 2 Gem depends on the following ROS 2 packages

```Cmake
target_depends_on_ros2_packages(${gem_name}.Static rclcpp builtin_interfaces std_msgs sensor_msgs nav_msgs tf2_ros ackermann_msgs gazebo_msgs)
target_depends_on_ros2_package(${gem_name}.Static control_toolbox 2.2.0 REQUIRED)
```

However, during my build attempt in Docker, I came across a potential fatal bug. Apparently, there is a conflict between ```rmw_cyclonedds_cpp``` and ```rmw_implementation``` where it appears to find two instances of ```rmw_implementation::rmw_implementation```.  Please see the attached console output. I tried looking up this error online but did not find anything that resembles this exact problem. The closest I found was this: https://github.com/ros2/rmw_implementation/issues/4

Reproducing this error is somewhat challenging as it happens only when compiling an O3DE project with ROS 2 Gem. To help reproduce this error I have created a public repository](https://github.com/Mechazo11/space_ros_rmw_conflict) which will build the ```rmw_conflict``` image based on ```osrf/space_ros``` base image.  Please allot 30GB space and 20-25mins to building the ```rmw_conflict``` image. After the image has been built, follow these steps

* Open a container: ```docker run -it rmw_conflict /bin/bash```

* Execute the following commands in the container's terminal

```bash
source /opt/ros/humble/setup.bash
source $SPACEROS_DIR/install/setup.bash
cmake -B build/linux -G "Ninja Multi-Config" -DLY_DISABLE_TEST_MODULES=ON -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DLY_STRIP_DEBUG_SYMBOLS=ON -DAZ_USE_PHYSX5:BOOL=ON
```
The error message shown in the provided console output clearly shows ```rmw_cyclonedds_cpp``` and ```rmw_implementation``` are in odds with each other.

I **was** able to solve this problem but it involved deleting the ```Install``` directory of the base spaceros workspace and rebuilding it. This was done in the following steps

```bash
# Clone all space-ros sources
RUN mkdir ${SPACEROS_DIR}/src \
  && vcs import ${SPACEROS_DIR}/src < ${SPACEROS_DIR}/exact.repos

RUN sudo sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" > /etc/apt/sources.list.d/ros2-latest.list'
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -

# Clear Install folder, we will need to rebuild the workspace again
WORKDIR ${SPACEROS_DIR}
RUN sudo rm -rf install/

# Delete problematic packages
WORKDIR ${SPACEROS_DIR}/src
RUN sudo rm -rf rmw_cyclonedds_cpp
RUN sudo rm -rf rmw_implementation  
RUN sudo rm -rf rmw_cyclonedds
RUN sudo rm -rf realtime_tools

WORKDIR ${SPACEROS_DIR}

# Update rosdep
RUN --mount=type=cache,target=/var/cache/apt,sharing=locked \
  --mount=type=cache,target=/var/lib/apt,sharing=locked \
  sudo apt-get update -y
RUN --mount=type=cache,target=/var/cache/apt,sharing=locked \
  --mount=type=cache,target=/var/lib/apt,sharing=locked \
  sudo apt-get dist-upgrade -y
RUN rosdep update

# Install them back from apt
RUN --mount=type=cache,target=/var/cache/apt,sharing=locked \
[... 146 lines cut ...]
  /home/spaceros-user/spaceros/install/libstatistics_collector/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/spaceros-user/spaceros/install/libstatistics_collector/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /home/spaceros-user/spaceros/install/rclcpp/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/spaceros-user/spaceros/install/rclcpp/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  /home/spaceros-user/spaceros/install/control_toolbox/share/control_toolbox/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/spaceros-user/spaceros/install/control_toolbox/share/control_toolbox/cmake/control_toolboxConfig.cmake:41 (include)
  /home/spaceros-user/spaceros/o3de-extras/Gems/ROS2/Code/ros2_target_depends.cmake:8 (find_package)
  /home/spaceros-user/spaceros/o3de-extras/Gems/ROS2/Code/CMakeLists.txt:73 (target_depends_on_ros2_package)

-- Found parameter_traits: 0.3.8 (/home/spaceros-user/spaceros/install/parameter_traits/share/parameter_traits/cmake)
-- Found rclcpp_lifecycle: 16.0.10 (/home/spaceros-user/spaceros/install/rclcpp_lifecycle/share/rclcpp_lifecycle/cmake)
-- Downloading package into /home/spaceros-user/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux
'/usr/bin/cmake' '-E' 'tar' 'xf' '/home/spaceros-user/.o3de/3rdParty/downloaded_packages/sdformat-13.5.0-rev2-linux.tar.xz'
-- Installed And Validated package at /home/spaceros-user/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux - OK
-- Using package /home/spaceros-user/.o3de/3rdParty/packages/sdformat-13.5.0-rev2-linux
-- Found geometry_msgs: 4.2.4 (/home/spaceros-user/spaceros/install/geometry_msgs/share/geometry_msgs/cmake)
-- Using all available rosidl_typesupport_c: rosidl_typesupport_introspection_c;rosidl_typesupport_fastrtps_c
-- Found rosidl_adapter: 3.1.5 (/home/spaceros-user/spaceros/install/rosidl_adapter/share/rosidl_adapter/cmake)
-- Using all available rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp;rosidl_typesupport_fastrtps_cpp
-- Downloading package into /home/spaceros-user/.o3de/3rdParty/packages/OpenMesh-8.1-rev3-linux
'/usr/bin/cmake' '-E' 'tar' 'xf' '/home/spaceros-user/.o3de/3rdParty/downloaded_packages/OpenMesh-8.1-rev3-linux.tar.xz'
-- Installed And Validated package at /home/spaceros-user/.o3de/3rdParty/packages/OpenMesh-8.1-rev3-linux - OK
-- Using package /home/spaceros-user/.o3de/3rdParty/packages/OpenMesh-8.1-rev3-linux
-- Configuring incomplete, errors occurred!
```

Mechazo11, 2024-09-22:
@Bckempa thank you very much for looking into this. Yes, I suspected this much why Space ROS preferred to source build. However at that time (and as of today), without the apt like patch, I cannot get O3DE's ROS 2 Gem to build which subsequently breaks the build. What can be the next step in troubleshooting this issue?

asimonov, 2024-09-22:
@Mechazo11, great writeup. i have to admit i did not read it entirely.

but can i suggest to try something to solve the original issue you face? not sure you have done this, but the following setup may be better:
* start with space-ros container
* source spaceros as underlay: source $SPACEROS_DIR/install/setup.bash
* create your workspace to put the packages that you need to build the Gem, using humble version of those. use rosinstall/vcstool/rosdep route as we do in spaceros containers, but check that rosdep does not bring 'apt install ros-xxx' packages (you can run rosdep with `-s` option to see what it is running
* build and source your workspace as overlay of spaceros
* build the gems and the rest of your application

Mechazo11, 2024-09-22:
Hi @asimonov 
No worries, after typing it up, even I was skeptical whether I would have read it myself. But I kept it nonetheless since the problem was confusing to me and caused me a lot of headache when I was trying to do my submission.

I recall trying this route by modifying the space_robots_demo container (```demo_ws```) but I will try it again by updating the docker file for the ```rmw_conflict``` image. Hopefully that will be more in line with Space ROS regulations and we can then proceed to finding out why the conflict occurs.

Mechazo11, 2024-10-08:
Hi @asimonov and @Bckempa , apologies for the long delay but over the last weekend I was able to follow along the suggested steps from the discussion we had a few weeks earlier. 

Even after ensuring I am not using any ros2-humble (```rosdep install``` installs a bunch, I remove all of them later on) like packages and correctly overlaying a ```colcon_ws``` on top of ```spaceros_ws``` only, I am still getting rmw conflict error. 

When convenient, can you kindly try out the following docker image to reproduce the error on your end. Please allot about 25GB of disk space.

* Instructions for building the image
```bash
docker pull osrf/space-ros
cd ~/Downloads
git clone https://github.com/Mechazo11/space_ros_rmw_conflict.git
cd space_ros_rmw_conflict/
docker build -t rmw_conflict -f Dockerfile_rmw_conflict .
```
* Open a docker container: ```docker run -it rmw_conflict /bin/bash```

* Execute the following commands
```bash
source ${SPACEROS_DIR}/install/setup.bash
source ${COLCON_WS_DIR}/install/setup.bash
cmake -B build/linux -G "Ninja Multi-Config" -DLY_DISABLE_TEST_MODULES=ON -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DLY_STRIP_DEBUG_SYMBOLS=ON -DAZ_USE_PHYSX5:BOOL=ON
```

Hopefully with some guidance, I should be able to finish this integration and have Open 3D Engine correctly supported in Space ROS.

asimonov, 2024-10-18:
I may try over the weekend.Does the build fail? Or runtime?_____________________________Kind Regards,Alexey Simonov

When convenient, would you kindly try out the docker file from the previous comment? I have tried to incorporate all the suggested steps but still saw the same problem. Maybe a fresh perspective would help us resolve this problem.
With best,
@Mechazo11

Mechazo11, 2024-10-18:
@asimonov yes you should see the build fail and reproduce the error shown in the first comment. From there is where I am confused on how to proceed in solving the error

Mechazo11, 2024-12-15:
For the time being I am closing off this ticket, will reopen it if I have some time to work on this problem in future.

closed 2024-12-15 by Mechazo11