space-ros/space-ros #218: ament_lint - space-ros branch update

asimonov, 2024-09-22:
### Feature description

we use dedicated `spaceros` branch of `ament_lint` (https://github.com/ament/ament_lint/tree/spaceros) which is 101 commits behind `rolling`

there is some ongoing work around ament_lint in space-ros. and also we will need to rebase that branch for Jazzy upgrade

last changes to spaceros branch of ament_lint are 2 years old, so it seems that this branch is not maintained. shall we clone it to space-ros org?

### Implementation considerations

* [ ] clone ament_lint to space-ros org
* [ ] change the build to reference space-ros version
* [ ] rebase ament_lint to current state of Humble
* [ ] apply on-going fixes for space-ros ament_lint
* [ ] create new jazzy branch and rebase to Jazzy ament_lint

asimonov, 2024-09-22:
I have tried to rebase ament_lint/spaceros to jazzy. it shows a few conflicts. I tried to resolve them, but got stuck as I am not familiar with underlying details of ament_lint or its space-ros usage.

May be Blazej can comment?

xfiderek, 2024-09-24:
I am on vacation till end of week so I can elaborate more on it next week, but in general I think the only reason we use a separate branch for ament lint is that ament_lint_common was patched there to include ament_cobra (but to be 2x checked). This way packages that include ament_lint_common automatically include cobra (and i think ikos) as linters. This approach is not bulletproof, as packages that do not use ament_lint_auto and ament_lint_common won't generate cobra and ikos reports. I proposed an alternative in this PR: https://github.com/space-ros/space-ros/pull/145. If indeed the only change to base branch was to include cobra and ikos, then i think this branch could be decommissioned after my PR is merged. This is something to investigate and i am happy to spend time on it next week.

asimonov, 2024-09-24:
Thanks Blazej! Have a nice vacation!
It can definitely wait, but it would be great to rework this arrangement with ament_lint to make things more straightforward. Looks like you have the background knowledge required.
I would have a look (even out of curiousity) but I do not have much time for few weeks :(

asimonov, 2024-10-08:
> I am on vacation till end of week so I can elaborate more on it next week, but in general I think the only reason we use a separate branch for ament lint is that ament_lint_common was patched there to include ament_cobra (but to be 2x checked). This way ROSpackages that leverage ament_lint_common automatically include cobra (and i think ikos) in spaceros as linters. This approach is not bulletproof, as packages that do not use ament_lint_auto and ament_lint_common won't generate cobra and ikos reports. I proposed an alternative in this PR: #145. If indeed the only change to base branch was to include cobra and ikos in ament_lint_common package.xml, then i think spaceros branch could be decommissioned after my PR is merged and we would simply use upstream tag (jazzy). This is something to investigate and i am happy to spend time on it next week.
> 
> EDIT: there is a PR open in ament_lint repo with some changes required to make #145 working, but it should be possible to merge it to base branch of ament_lint without major issues

@xfiderek I am coming back to this after couple of hectic weeks.

we can do PR #145 as you propose, but that is pending merge ament_lint#482 which was going nowhere since march. if it is merged and we can just just upstream ament_lint that would be awesome. but what if this is not happening? shall we still bring ament_lint space-ros branch to space-ros org?

on a wider topic, in our ros2.repos we have exactly 4 un-pinned repos:

```
  ament_cobra:
    type: git
    url: https://github.com/ament/ament_cobra.git
    version: master

  ament_ikos:
    type: git
    url: https://github.com/ament/ament_ikos.git
    version: main

  ament_lint:
    type: git
    url: https://github.com/ament/ament_lint.git
    version: spaceros

  process_sarif:
    type: git
    url: https://github.com/space-ros/process_sarif.git
    version: main

```

process_sarif is already in space-ros. 
we ideally want to reference tags on the other 3.
ok, ament_lint is ROS2-wide, so better merge our PR there and just reference upstream.
What about ament_cobra and ament_ikos? They seem to be space-ros specific. Shall we move them over to space-ros to be able to tag them at will when we do releases?

asimonov, 2024-10-18:
> I am on vacation till end of week so I can elaborate more on it next week, but in general I think the only reason we use a separate branch for ament lint is that ament_lint_common was patched there [...]

@xfiderek , I have updated the issue description to make it open for proposals as to how to move ament_lint to jazzy as part of space-ros.

given that ament_lint PR that you mention is merged, can you check if #145 would indeed allow us to simply switch to current jazzy branch of ament_lint, ignoring 22 commits on spaceros branch?

asimonov, 2024-10-18:
I just tried to run the tests again, this time dropping into interactive session of Earthly and running them for just ament_lint. I can see that @xfiderek latest changes are there, so it is definitely picking up the right branch. but it fails for some reason which is not reported:

```
spaceros-user@buildkitsandbox:/opt/spaceros/src/ament_lint$ git log
commit 9fe34febc580eb943a2f64d19aec2d74ef70021e (grafted, HEAD)
Author: Błażej Fiderek <57559956+xfiderek@users.noreply.github.com>
Date:   Tue Oct 15 20:36:17 2024 +0200

    Add AMENT_LINT_AUTO_SKIP_PREEXISTING_TESTS flag (#482)
    
    Signed-off-by: xfiderek <fiderekblazej@gmail.com>
spaceros-user@buildkitsandbox:/opt/spaceros/src/ament_lint$ cd ../..
spaceros-user@buildkitsandbox:/opt/spaceros$ colcon test  --packages-up-to ament_lint
Starting >>> ament_lint
--- stderr: ament_lint                   

----------------------------------------------------------------------
Ran 0 tests in 0.000s

NO TESTS RAN
---
Failed   <<< ament_lint [0.25s, exited with code 5]

Summary: 0 packages finished [0.78s]
  1 package failed: ament_lint
  1 package had stderr output: ament_lint
spaceros-user@buildkitsandbox:/opt/spaceros$ 
```

xfiderek, 2024-10-22:
Hi @asimonov,

Upon reviewing the ament_lint issue, I’ve arrived at the same conclusions you did. To address the issue on SpaceROS Jazzy, I identified that we would need to back-port the following commit: https://github.com/ament/ament_lint/commit/903c28344c5fbe787c5882d4ba255bf220277cd9.

After comparing the jazzy and spaceros branches of ament_lint, I noticed that several features have been added to the spaceros branch that cannot be merged easily without further testing (e.g., SARIF support across all packages and MISRA compliance in ament_cppcheck).

While I’m happy to review and test these changes, **given the limited time before the release**, my recommendation for this cycle is to back-port or cherry-pick the mentioned commit to the spaceros branch. This will resolve the immediate ament_lint issue without introducing additional complexity.

Let me know, @ivanperez-keera, if this approach works for you as well.

In the long term, I believe we can merge the additional features from the spaceros branch into rolling to eliminate the need for a fork and simplify future maintenance.

Looking forward to your input, @asimonov and @ivanperez-keera.

asimonov, 2024-10-22:
I dont have a strong view.

but if we backport some commits to current spaceros branch of ament_lint - will main/humble space-ros repo still work? Or we will have 2 ament_lint branches to maintain?

xfiderek, 2024-10-22:
It *should* work, but for sure I will test it before merging

asimonov, 2024-10-29:
discussed this offline with @xfiderek and Ivan.

The long-term plan is to merge some of 22 commits on `spaceros` `ament_lint` branch to `rolling`. Meanwhile we will maintain `spaceros-main` branch on ament_lint and convert current spaceros branch to a tag to preserve for past spaceros releases. Hopefully we can create spaceros-release-xxxx.yy tags on ament_lint, but if not possible then may be temporarily we need to use a spaceros org fork of ament_lint until the above commits are fully upstreamed?

for [imminent] jazzy release of spaceros @xfiderek may backport unittest fix for ament_lint so we do not exclude this package from build-testing. There are no tests in ament_lint anyway, so it is not too important, but would be good to not exclude packages.