space-ros/space-ros #219: Ensure the latest main image is available for downstream consumption

eholum, 2024-09-22:
### Feature description

Right now we push the latest `main` build of the space ros base image to the GHCR. Unfortunately, the project permissions prevent anyone except for this repository and org members from pulling the image at:

`ghcr.io/space-ros/space-ros:main`.

Not even builds from other projects in the organization [have access to the pushed images](https://github.com/space-ros/docker/actions/runs/10941929534/job/30377716780#step:6:197).

To unblock CI in the docker project, I recommend that we either modify the package permissions for this project's docker images to make them public, or push the main images to Dockerhub under the tag, `osrf/space-ros:main`, or both.

### Implementation considerations

In the interest of expediency, I suggest adding a step to push the `main` image to dockerhub as well, and follow up with registry permissions later.

closed 2024-09-26 by eholum

asimonov, 2024-09-27:
woo hoo, it worked!
https://hub.docker.com/layers/osrf/space-ros/main/images/sha256-56db8c53ed27f1648cec50261ff24ee8fadd25aefa9835d3850d46342a4171eb?context=explore

eholum, 2024-09-27:
Progress!