space-ros/space-ros #221: does earthly trigger matrix build?

asimonov, 2024-09-24:
### Version or commit hash

main

### RMW Implementation

CycloneDDS

### Steps to reproduce issue

just trigger "earthly +sources"

[image]

### Expected behavior

expecting 'spaceros-artifacts' and other steps to be triggered just once?

### Actual behavior

'spaceros-artifacts' stage is run at least 3 times in parallel, see the picture.

but also in the CI: https://github.com/space-ros/space-ros/actions/runs/11015522237/job/30588788505?pr=220#step:5:3388 (notice vcs import is run 3 times...)

it looks like Earthly is doing 'matrix build' (https://docs.earthly.dev/docs/guides/build-args#matrix-builds), but why? Is it wrong behaviour?

asimonov, 2024-09-24:
one side effect of this 'vcs import' triggered multiple times is that some network infrastructure (crowdstrike tier?) starts to refuse more 'git clones', resulting in cloning errors:

[image]

asimonov, 2024-09-24:
upon further investigation the culprit is "RUN --no-cache".
If I remove '--no-cache' then only one job gets to run:

[image]

so, i suggest we remove "--no-cache" options as they result in this strange behaviour.

xfiderek, 2024-10-02:
We need to be careful about removing one of the statements. it may require longer brainstorming. Please see the PR comment https://github.com/space-ros/space-ros/pull/222/files#r1784844809