space-ros/space-ros #225: Improve/Refactor Earthly CI build workflow

franklinselva, 2024-10-17:
### Feature description

The current earthly CI build workflow runs sequentially with the build tests being run daily. This takes up approximately 2.45 hrs for the build tests to complete each day. I am not sure of the build minutes allocated for the organization. I think the time window for the scheduled earthly build test can be extended. Additionally, the image being pushed to the gihub registry and dockerhub can be performed in parallel to save some build minutes.

### Implementation considerations

A few improvements can be performed in my opinion,
 - The scheduled jobs being run on the Earthly build workflow can be lenient. One of the considerations depends on the [ros2.repos](https://github.com/space-ros/space-ros/blob/main/ros2.repos) which are mostly versioned (except ament_cobra, ament_lint, ament_ikos & process_sarif which refers to a branch). 
 - Parallel CI build and pushing the images to registry. The flow can look something like this.

```mermaid
graph LR
  J[Start] --> A[Build Test] 
  J --> B[Build Core Image]
  J --> C[Build Dev Image]
  A --> D[Push Core Image to GHCR]
  A --> E[Push Dev Image to GHCR]
  A --> F[Push Core Image to Dockerhub]
  A --> G[Push Dev Image to Dockerhub]
  B --> D
  B --> E
    C --> F
    C --> G

subgraph Preliminary Jobs
A
B
C
end

subgraph Push Images
D
E
F
G
end
```

ivanperez-keera, 2025-05-08:
Is this already covered by https://github.com/space-ros/space-ros/issues/159 and PR 186?

eholum, 2025-07-19:
Another TODO for this: Add a concurrency check.

ivanperez-keera, 2026-10-01:
Hi,

Would this increase the build times where `test` wouldn't start until `core` and `dev` finish?

This issue has been stagnant for over a year. If we are not able to make progress on this and it requires more active discussion, I propose we move this to a discussion to keep our issue list organized and going in a consistent direction.

ivanperez-keera, 2026-10-19:
I'm a little bit confused by this and by #430.

Does #430 cover everything that is being discussed here?

[It sounds like the concurrency check was a side todo](https://github.com/space-ros/space-ros/issues/225#issuecomment-3092413332) not the main issue.

asimonov, 2026-10-20:
this is the old original issue for refactoring image builds.

it still says 'earthly' which is now replaced by docker.

given all the work on improving build infrastructure recently this whole issue need to be reviewed or closed/distilled to remaining improvements