space-ros/space-ros #235: add push to docker hub for jazzy branch

asimonov, 2024-11-28:
### Feature description

To proceed with demos porting to Jazzy we need a readily available Jazzy based image.

We can either:
* merge jazzy to main or 
* modify workflow to push jazzy tagged image to docker hub when jazzy branch is built

### Implementation considerations

* The github 'cron' action is only running on main.
* I suggest to rely on manual workflow run on jazzy branch for now which should push jazzy tag to docker hub

EzraBrooks, 2024-12-02:
I'm in favor of merging jazzy to main because I doubt Space ROS will ever track rolling, at least in the near term

asimonov, 2024-12-03:
lets merge jazzy then!

EzraBrooks, 2024-12-12:
Decision: merge jazzy to main, but make sure to produce an explainer document about why we are moving forward with newer OSes and how to use Space ROS on older ones.

Mechazo11, 2024-12-16:
Hi @EzraBrooks I had a quick question. I am interested in helping Space ROS natively integrate with Harmonic. When this upcoming merge happens, is it only the base image that will be upgraded?

Mechazo11, 2024-12-16:
Hi @Bckempa apologies for the confusion. In regards to your question, I "sort of" understood whats going on. Is the plan is then to upgrade the base image to ```jazzy``` and then add additional overlays to roll out support for Harmonic, Moveit2 etc. depending on which demo is required to be built?

Mechazo11, 2024-12-16:
Yes thank you very much @Bckempa. After the base image gets upgraded, can I put forward a proposal on how to add Gazebo Harmonic with Space ROS?

asimonov, 2024-12-20:
> Decision: merge jazzy to main, but make sure to produce an explainer document about why we are moving forward with newer OSes and how to use Space ROS on older ones.

On a practical note - how should this be done? a new issue with request to merge jazzy to main plus explainer document?

asimonov, 2025-02-03:
shall we cancel this?

now main is jazzy based, so no need for separate docker image to be pushed

asimonov, 2025-03-30:
closing this as jazzy was merged to main

closed 2025-03-30 by asimonov