space-ros/space-ros #294: IKOS build fails

SEC-RKatayama, 2025-07-20:
### Version or commit hash

jazzy-2025.04.0

### RMW Implementation

FastRTPS

### Steps to reproduce issue

Just follow USAGE.md, and it happens when running colcon build command at ["Running an IKOS Scan" section](https://github.com/space-ros/space-ros/blob/jazzy-2025.04.0/docs/USAGE.md#running-an-ikos-scan).
(With `--packages-select rcpputils`)

```bash
CC="ikos-scan-cc" CXX="ikos-scan-c++" LD="ikos-scan-cc" colcon build --build-base build_ikos --install-base install_ikos --packages-select rcpputils --cmake-args -DSECURITY=ON -DINSTALL_EXAMPLES=OFF -DCMAKE_EXPORT_COMPILE_COMMANDS=ON --no-warn-unused-cli

```

Environment:
- WSL2 on Windows11
- Ubuntu-24.04

### Expected behavior

Succeed in colcon build.

### Actual behavior

It seems that `ikos-scan-cc` failed to compile or link.

The error message comes from L.67 in `/usr/share/cmake-3.28/Modules/CMakeTestCCompiler.cmake` below.

```cmake
  try_compile(CMAKE_C_COMPILER_WORKS
    SOURCE_FROM_VAR testCCompiler.c __TestCompiler_testCCompilerSource
    OUTPUT_VARIABLE __CMAKE_C_COMPILER_OUTPUT)
  unset(__TestCompiler_testCCompilerSource)
  # Move result from cache to normal variable.
  set(CMAKE_C_COMPILER_WORKS ${CMAKE_C_COMPILER_WORKS})
  unset(CMAKE_C_COMPILER_WORKS CACHE)
  __TestCompiler_restoreTryCompileTargetType()
  if(NOT CMAKE_C_COMPILER_WORKS)
    PrintTestCompilerResult(CHECK_FAIL "broken")
    string(REPLACE "\n" "\n  " _output "${__CMAKE_C_COMPILER_OUTPUT}")
    message(FATAL_ERROR "The C compiler\n  \"${CMAKE_C_COMPILER}\"\n"
      "is not able to compile a simple test program.\nIt fails "
      "with the following output:\n  ${_output}\n\n"
      "CMake will not be able to correctly generate this project.")
```

### Backtrace or Console output

```log
[1.883s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
        'rcpputils' is in: /opt/spaceros/install/rcpputils
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
        --allow-overriding rcpputils

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> rcpputils
[1.961s] WARNING:colcon.colcon_core.shell:The following packages are in the workspace but haven't been built:
- ament_cppcheck
- ament_lint
- ament_package
- google_benchmark_vendor
- gtest_vendor
- osrf_pycommon
[... 142 lines cut ...]

    Run Build Command(s): /usr/bin/cmake -E env VERBOSE=1 /usr/bin/gmake -f Makefile cmTC_50bb0/fast
    /usr/bin/gmake  -f CMakeFiles/cmTC_50bb0.dir/build.make CMakeFiles/cmTC_50bb0.dir/build
    gmake[1]: Entering directory '/opt/spaceros/build_ikos/rcpputils/CMakeFiles/CMakeScratch/TryCompile-8RlQ3s'
    Building C object CMakeFiles/cmTC_50bb0.dir/testCCompiler.c.o
    /opt/ikos/bin/ikos-scan-cc    -MD -MT CMakeFiles/cmTC_50bb0.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_50bb0.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_50bb0.dir/testCCompiler.c.o -c /opt/spaceros/build_ikos/rcpputils/CMakeFiles/CMakeScratch/TryCompile-8RlQ3s/testCCompiler.c
    Linking C executable cmTC_50bb0
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_50bb0.dir/link.txt --verbose=1
    /opt/ikos/bin/ikos-scan-cc CMakeFiles/cmTC_50bb0.dir/testCCompiler.c.o -o cmTC_50bb0
    error: file: No such file or directorygmake[1]: *** [CMakeFiles/cmTC_50bb0.dir/build.make:100: cmTC_50bb0] Error 2
    gmake[1]: *** Deleting file 'cmTC_50bb0'
    gmake[1]: Leaving directory '/opt/spaceros/build_ikos/rcpputils/CMakeFiles/CMakeScratch/TryCompile-8RlQ3s'
    gmake: *** [Makefile:127: cmTC_50bb0/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

---
Failed   <<< rcpputils [7.27s, exited with code 1]

Summary: 0 packages finished [8.96s]
  1 package failed: rcpputils
  1 package had stderr output: rcpputils
```

mkhansenbot, 2025-07-27:
We just updated the docs on this, can you take a look at https://github.com/space-ros/space-ros/blob/main/docs/IKOS.md. If that doesn't work, please update this issue.

SEC-RKatayama, 2025-07-29:
Thank you.
It does works (on main branch)!

[image]

---
`cobra_vendor` package fails to build, but I think it's intended, isn't it.

[image]

mkhansenbot, 2025-07-30:
@ivanperez-keera - can you look at the cobra error above?

ivanperez-keera, 2026-02-02:
Based on the conversation above, it looks like the issue with IKOS is sorted, so I'm closing this.